Klipper3d/klipper #1660: Unexpected out of range move on M25/M24

archer183, 2019-05-20:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3198267/klippy.log)

Occasionally (not all the time, generally only on longer running print) when M25 to pause is issued (I am using virtual SD card) at some point either after that command or upon M24, an out of range move is commanded.

Most of the time, however (and never with short print jobs) M25 simply pauses where it is, and M24 resumes properly.  I have not been able to replicate on short print jobs, only on ones that are long (and even then not all the time)

KevinOConnor, 2019-05-25:
This sounds like a gcode state mismatch - the gcode in the file issues one or more of M82/M83/G90/G91/G92 commands to set the gcode parsing state.  During the pause, something else changes the parsing state via M82/M83/G90/G91/G92 and then when a resume is attempted the incorrect state causes movement commands to have different meanings.  This is a fundamental limitation of g-code.

-Kevin

KevinOConnor, 2019-05-25:
FWIW, the log does show what the g-code state was at the time of the M112:
```
gcode state: absolutecoord=True absoluteextrude=False base_position=[0.0, 0.0, 0.0, 20.0] last_position=[240.0, 240.0, 138.6, 37256.075169993484] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.0166666666667 extrude_factor=1.0 speed=500.0
```
If you have access to the g-code print that failed, you could check if that state is indeed in conflict with the desired state.

-Kevin

archer183, 2019-05-26:
I will see if I can replicate it on a short run and document all steps.  it was really strange as it was obvious that some code move was commanded after the pause, even though I did not issue any command (I.E.  normal pause, head stops immediately where it is, abnormal, it immediately retracts from part)    Now that I look back, the start of this oddly coincides with the latest octoprint update being installed...

archer183, 2019-05-31:
I was able to replicate (after many tries) using the following steps.

**Faulty response / procedure is below**
using klipper menu interface to start gcode run from virtual SD card
Let run for a while.   
Close octoprint tab in web browser connected to 192.168.2.83 (the printer IP)
using klipper menu interface in sd menu, select pause  - does not pause run, does nothing
Re-open octoprint in web browser
issue M25 from octoprint terminal -  Pause occurs but immediately afterwards, an uncommanded Z retract occurs.  If this happens, the issue has occurred as normally issuing M25 causes the print to pause but make no further moves.
issue M24 from octoprint terminal to resume 

get error in octoprint "Your printer's firmware reported an error. Due to that the ongoing print job will be cancelled. Reported error: Move out of range: 311.583 274.186 48.300 [5758.772]"

log file
[klippy.log](https://github.com/KevinOConnor/klipper/files/3239963/klippy.log)

copy of octoprint terminal output
[octoprint terminal output.txt](https://github.com/KevinOConnor/klipper/files/3239993/octoprint.terminal.output.txt)

gcode being run from SD card
[CFFFP_3DBenchy.zip](https://github.com/KevinOConnor/klipper/files/3239958/CFFFP_3DBenchy.zip)

KevinOConnor, 2019-05-31:
After running the log through `scripts/logextract.py` I get:
```
Read 5880.415461: 'M25\n'
Read 5880.417152: 'M25\n'
...
Exiting SD card print (position 4837143)
Read 5880.544522: 'M400\n'
...
Read 5883.578697: 'M114\n'
Read 5883.611341: 'M27\n'
Read 5883.618163: 'G91\n'
Read 5883.626652: 'G1 E-10.0\n'
Read 5883.627914: 'G1 Z+10.0\n'
...
Starting SD card print (position 4837143)
Move out of range: 311.583 274.186 48.300 [5758.772]
Exiting SD card print (position 4837143)
...
Read 5888.559092: 'M24\n'
```

Not sure why OctoPrint sent a G91 (or the retract+lift), but that's what's killing your print.  (It commanded the gcode mode into relative mode, while the print likely required absolute mode.)  I'd check the OctoPrint scripts (wrench->GCODE Scripts) to see if there is something in there.

FWIW, the SAVE_GCODE_STATE/RESTORE_GCODE_STATE commands were recently added.  It should be possible to make macros that pause+save and then restore+resume to try and avoid this issue.

-Kevin

archer183, 2019-06-01:
it is interesting.  is there any reason why the klipper menu command for pause would stop working?   (because it does not work when this fault occurs)  the whole reason I have to issue it from octoprint is because the menu sometimes doesn't send the command.  It seems to be linked to shutting down the browser connected to octoprint, as I have never seen the fault with the browser open the whole print.  

found code source in octoprint execute after pause (now removed), but it only executes this if I have closed the original browser and reopened it later.  if I do not close the browser, M25 works from terminal properly.  The  other thing I noticed is that the default klipper menu "pause" option under the sd card menu does nothing under any state.    I could have sworn it worked at some point, but maybe I am mistaken.

KevinOConnor, 2019-06-01:
> is there any reason why the klipper menu command for pause would stop working? 

Issue #1529.  (The menu continually gives g-code priority to virtual_sdcard.)

-Kevin

KevinOConnor, 2019-06-18:
Any further updates on this?

-Kevin

archer183, 2019-06-18:
Not really.   I still see inconsistent action from pause when running
virtual sd.  Sometimes it executes the octoprint after pause gcode,
sometimes it does not.

KevinOConnor, 2019-06-18:
Are you on the latest version of the code?  If you are using the pause_resume module, then be sure you have commit 05a04d5d.

-Kevin

archer183, 2019-06-21:
I am now on whatever pulls down as of last night.  I will attempt some testing this weekend.

archer183, 2019-06-23:
I have been unable so far to replicate the fault on the newest available klipper and the buttons for pause and resume appear to work.

KevinOConnor, 2019-07-10:
Okay - it sounds like we can close this issue then.

-Kevin

closed 2019-07-10 by KevinOConnor