Klipper3d/klipper #1663: homing_positive_dir suddenly doesn´t work

Xkc0, 2019-05-21:
Hey
I was setting up the homing on my Anet A6 with an MKS Gen l Board and at first, the x- and y-axis homed just fine with _homing_positive_dir: false_ but after setting up the homing override for my BLtouch the y-axis wasn´t homing anymore towards the back. I tried to uncomend the homing override again but that didn´t fix the issue. 
What can I do to fix this?

Here is my config file and the Log
Thanks and awesome piece of Software you created here. 

[Config_MKS Gen L.txt](https://github.com/KevinOConnor/klipper/files/3204050/Config_MKS.Gen.L.txt)

[klippy.log](https://github.com/KevinOConnor/klipper/files/3204028/klippy.log)

james-golden, 2019-05-22:
Have you dried to remove the “!” In front of the dir pin for you Y stepper?

Xkc0, 2019-05-22:
Not yet. I´m going to try that tomorrow. 
Thanks for the reply

Xkc0, 2019-05-23:
That solved it.
Thanks a lot.

closed 2019-05-23 by Xkc0