Klipper3d/klipper #1664: MAX6675 Temperature is wrong

Geramy, 2019-05-22:
at 235C reading in klipper my Multimeter is reading the thermocouple as 80C also I can tell this because at 235C I should be able to at least print the PLA some what it won't even become hot enoough to partially melt the PLA I can literally keep my finger on the extruder nozzle for over 3 seconds at the "so called" 235C reading. Its incorrect. Smoothieware was able to read this right but klipper is not reading the MAX6675 correctly.

**********************************LOG***********************************
Start printer at Wed May 22 07:33:54 2019 (1558506834.5 310.6)
===== Config file =====
[tmc2130 stepper_x]
cs_pin = P1.17
spi_software_sclk_pin = P0.4
spi_software_mosi_pin = P4.28
spi_software_miso_pin = P0.5
microsteps = 16
run_current = 1.000
hold_current = 0.900
stealthchop_threshold = 150

[tmc2130 stepper_y]
cs_pin = P1.15
spi_software_sclk_pin = P0.4
spi_software_mosi_pin = P4.28
spi_software_miso_pin = P0.5
microsteps = 16
run_current = 1.000
hold_current = 0.900
stealthchop_threshold = 250

[tmc2130 stepper_z]
cs_pin = P1.10
spi_software_sclk_pin = P0.4
spi_software_mosi_pin = P4.28
spi_software_miso_pin = P0.5
microsteps = 16
run_current = 1.300
hold_current = 0.900
stealthchop_threshold = 30

[tmc2130 extruder]
cs_pin = P1.8
spi_software_sclk_pin = P0.4
spi_software_mosi_pin = P4.28
spi_software_miso_pin = P0.5
microsteps = 16
run_current = 1.000
hold_current = 0.900
stealthchop_threshold = 5

[tmc2130 extruder1]
cs_pin = P1.1
spi_software_sclk_pin = P0.4
spi_software_mosi_pin = P4.28
spi_software_miso_pin = P0.5
microsteps = 16
run_current = 1.000
hold_current = 0.800
stealthchop_threshold = 5

[stepper_x]
step_pin = P2.2
dir_pin = P2.6
enable_pin = !P2.1
step_distance = .0125
endstop_pin = !P1.28
position_endstop = -120
position_max = 110
position_min = -120
homing_speed = 50
homing_positive_dir = False

[stepper_y]
step_pin = P0.19
dir_pin = P0.20
[... 518 lines cut ...]
Stats 748.7: gcodein=127 mcu: mcu_awake=0.002 mcu_task_avg=0.000009 mcu_task_stddev=0.000012 bytes_write=4497 bytes_read=68997 bytes_retransmit=9 bytes_invalid=0 send_seq=605 receive_seq=605 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001589 heater_bed: target=0 temp=37.4 pwm=0.000 print_time=394.210 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 extruder1: target=0 temp=236.5 pwm=0.000
Stats 749.7: gcodein=132 mcu: mcu_awake=0.002 mcu_task_avg=0.000009 mcu_task_stddev=0.000012 bytes_write=4503 bytes_read=69174 bytes_retransmit=9 bytes_invalid=0 send_seq=606 receive_seq=606 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001588 heater_bed: target=0 temp=37.3 pwm=0.000 print_time=394.210 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 extruder1: target=0 temp=237.0 pwm=0.000
Stats 750.7: gcodein=132 mcu: mcu_awake=0.002 mcu_task_avg=0.000009 mcu_task_stddev=0.000012 bytes_write=4509 bytes_read=69337 bytes_retransmit=9 bytes_invalid=0 send_seq=607 receive_seq=607 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001588 heater_bed: target=0 temp=37.3 pwm=0.000 print_time=394.210 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 extruder1: target=0 temp=238.0 pwm=0.000
Stats 751.7: gcodein=132 mcu: mcu_awake=0.002 mcu_task_avg=0.000009 mcu_task_stddev=0.000012 bytes_write=4515 bytes_read=69485 bytes_retransmit=9 bytes_invalid=0 send_seq=608 receive_seq=608 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001590 heater_bed: target=0 temp=37.2 pwm=0.000 print_time=394.210 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 extruder1: target=0 temp=238.5 pwm=0.000
Stats 752.7: gcodein=132 mcu: mcu_awake=0.002 mcu_task_avg=0.000009 mcu_task_stddev=0.000012 bytes_write=4521 bytes_read=69677 bytes_retransmit=9 bytes_invalid=0 send_seq=609 receive_seq=609 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001590 heater_bed: target=0 temp=37.2 pwm=0.000 print_time=394.210 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 extruder1: target=0 temp=238.5 pwm=0.000
Stats 753.7: gcodein=132 mcu: mcu_awake=0.002 mcu_task_avg=0.000009 mcu_task_stddev=0.000012 bytes_write=4527 bytes_read=69840 bytes_retransmit=9 bytes_invalid=0 send_seq=610 receive_seq=610 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001589 heater_bed: target=0 temp=37.2 pwm=0.000 print_time=394.210 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 extruder1: target=0 temp=238.5 pwm=0.000
Stats 754.7: gcodein=137 mcu: mcu_awake=0.002 mcu_task_avg=0.000009 mcu_task_stddev=0.000012 bytes_write=4533 bytes_read=69988 bytes_retransmit=9 bytes_invalid=0 send_seq=611 receive_seq=611 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001588 heater_bed: target=0 temp=37.2 pwm=0.000 print_time=394.210 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 extruder1: target=0 temp=239.0 pwm=0.000
Stats 755.7: gcodein=137 mcu: mcu_awake=0.002 mcu_task_avg=0.000009 mcu_task_stddev=0.000012 bytes_write=4539 bytes_read=70165 bytes_retransmit=9 bytes_invalid=0 send_seq=612 receive_seq=612 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001589 heater_bed: target=0 temp=37.2 pwm=0.000 print_time=394.210 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 extruder1: target=0 temp=238.5 pwm=0.000
Stats 756.7: gcodein=137 mcu: mcu_awake=0.002 mcu_task_avg=0.000009 mcu_task_stddev=0.000012 bytes_write=4545 bytes_read=70328 bytes_retransmit=9 bytes_invalid=0 send_seq=613 receive_seq=613 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001590 heater_bed: target=0 temp=37.2 pwm=0.000 print_time=394.210 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 extruder1: target=0 temp=238.0 pwm=0.000
Stats 757.7: gcodein=137 mcu: mcu_awake=0.002 mcu_task_avg=0.000009 mcu_task_stddev=0.000012 bytes_write=4551 bytes_read=70491 bytes_retransmit=9 bytes_invalid=0 send_seq=614 receive_seq=614 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001591 heater_bed: target=0 temp=37.1 pwm=0.000 print_time=394.210 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 extruder1: target=0 temp=238.0 pwm=0.000
Stats 758.7: gcodein=137 mcu: mcu_awake=0.002 mcu_task_avg=0.000009 mcu_task_stddev=0.000012 bytes_write=4557 bytes_read=70683 bytes_retransmit=9 bytes_invalid=0 send_seq=615 receive_seq=615 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001591 heater_bed: target=0 temp=37.1 pwm=0.000 print_time=394.210 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 extruder1: target=0 temp=238.0 pwm=0.000
Stats 759.7: gcodein=142 mcu: mcu_awake=0.002 mcu_task_avg=0.000009 mcu_task_stddev=0.000012 bytes_write=4563 bytes_read=70831 bytes_retransmit=9 bytes_invalid=0 send_seq=616 receive_seq=616 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001591 heater_bed: target=0 temp=37.1 pwm=0.000 print_time=394.210 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 extruder1: target=0 temp=237.0 pwm=0.000
Stats 760.7: gcodein=142 mcu: mcu_awake=0.002 mcu_task_avg=0.000009 mcu_task_stddev=0.000012 bytes_write=4569 bytes_read=70979 bytes_retransmit=9 bytes_invalid=0 send_seq=617 receive_seq=617 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001596 heater_bed: target=0 temp=37.1 pwm=0.000 print_time=394.210 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 extruder1: target=0 temp=237.0 pwm=0.000
Stats 761.7: gcodein=142 mcu: mcu_awake=0.002 mcu_task_avg=0.000009 mcu_task_stddev=0.000012 bytes_write=4575 bytes_read=71171 bytes_retransmit=9 bytes_invalid=0 send_seq=618 receive_seq=618 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001595 heater_bed: target=0 temp=37.1 pwm=0.000 print_time=394.210 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 extruder1: target=0 temp=236.0 pwm=0.000
[... cut ...] target=0 temp=0.0 pwm=0.000 extruder1: target=0 temp=230.0 pwm=0.000
Stats 772.8: gcodein=152 mcu: mcu_awake=0.002 mcu_task_avg=0.000009 mcu_task_stddev=0.000012 bytes_write=4647 bytes_read=72992 bytes_retransmit=9 bytes_invalid=0 send_seq=630 receive_seq=630 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100001592 heater_bed: target=0 temp=36.9 pwm=0.000 print_time=394.210 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 extruder1: tar

Geramy, 2019-05-22:
[klipper.log](https://github.com/KevinOConnor/klipper/files/3206139/klipper.log)

KevinOConnor, 2019-05-22:
The software spi mechanism was only recently added.  I suspect that system isn't working well for the MAX6675.  As a test, would you be able to hook up one of the thermocouples to the hardware spi pins (miso,mosi,sclk = P0.17,P0.18,P0.15) and check if the same issue occurs when using hardware spi?

-Kevin

KevinOConnor, 2019-05-22:
On a closer inspection, it looks like the MAX6675 and MAX31855 should be using spi mode 0 instead of mode 1.  Can you grab the latest code and retry?

See the FAQ for upgrade instructions.

-Kevin

Geramy, 2019-05-22:
Kevin, thank you, I can do that when I get home this evening.
Also how hard would it be to add support to use the GPIO on the rapsberrypi for things like SPI?
This is something i'm interested in working on.

KevinOConnor, 2019-05-22:
> Also how hard would it be to add support to use the GPIO on the rapsberrypi for things like SPI?
This is something i'm interested in working on.

If you're using hardware spi on the rpi, then I think it is just sysadmin and config work to launch the linux micro-controller code.  (See the beaglebone installation scripts for details on how to launch the linux code.)  If you want to use generic gpio on the rpi then it would require enhancing the src/linux/ code to support rpi gpio manipulation.

-Kevin

Geramy, 2019-05-22:
I was thinking I could make a C++ program with custom launch code for that app and allow python access to bit-banging the gpio ports. one shared core for communications and one dedicated core for gpio access. I believe the interrupt issue has been patched which was the only problem with getting bit banging to actually work properly. it was causing a 5 ms delay every second or two.

KevinOConnor, 2019-05-22:
The Klipper host code is organized to generate a schedule of events, and the Klipper micro-controller code is organized to run that schedule at the provided times.  I would recommend against trying to execute events directly from the host code.

-Kevin

Geramy, 2019-05-23:
I did the upgrade and now this is the results:
Recv: ok B:23.0 /0.0 T0:0.0 /0.0 T1:-99.6 /0.0

Geramy, 2019-05-23:
ITS WORKING!!!! Hey so my other thermocouple is not reading any temperature? Is there something wrong with the pins im using?

Geramy, 2019-05-23:
New small bug after printing something or well starting too:
Changing monitoring state from "Error: Heater extruder1 not heating at expected rate" to "Offline (Error: Heater extruder1 not heating at expected rate)"
Connection closed, closing down monitor

Recv: B:32.6 /60.0 T0:0.0 /0.0 T1:56.4 /295.0
Recv: B:32.9 /60.0 T0:0.0 /0.0 T1:58.2 /295.0
Recv: B:33.1 /60.0 T0:0.0 /0.0 T1:60.0 /295.0
Recv: B:33.5 /60.0 T0:0.0 /0.0 T1:62.4 /295.0
Recv: B:33.7 /60.0 T0:0.0 /0.0 T1:64.2 /295.0
Recv: B:34.0 /60.0 T0:0.0 /0.0 T1:66.1 /295.0
Recv: B:34.3 /60.0 T0:0.0 /0.0 T1:68.5 /295.0
Recv: B:34.5 /60.0 T0:0.0 /0.0 T1:70.3 /295.0
Recv: B:34.8 /60.0 T0:0.0 /0.0 T1:72.1 /295.0
Recv: B:35.1 /60.0 T0:0.0 /0.0 T1:74.4 /295.0
Recv: B:35.3 /60.0 T0:0.0 /0.0 T1:76.3 /295.0
Recv: B:35.6 /60.0 T0:0.0 /0.0 T1:78.1 /295.0
Recv: B:35.9 /60.0 T0:0.0 /0.0 T1:80.5 /295.0
Recv: B:36.1 /60.0 T0:0.0 /0.0 T1:82.3 /295.0
Recv: B:36.4 /60.0 T0:0.0 /0.0 T1:84.2 /295.0
Recv: B:36.7 /60.0 T0:0.0 /0.0 T1:86.7 /295.0
Recv: B:36.9 /60.0 T0:0.0 /0.0 T1:88.4 /295.0
Recv: B:37.2 /60.0 T0:0.0 /0.0 T1:90.2 /295.0
Recv: B:37.5 /60.0 T0:0.0 /0.0 T1:92.5 /295.0
Recv: B:37.7 /60.0 T0:0.0 /0.0 T1:94.2 /295.0
Recv: B:37.9 /60.0 T0:0.0 /0.0 T1:95.8 /295.0
Recv: B:38.2 /60.0 T0:0.0 /0.0 T1:98.1 /295.0
Recv: B:38.4 /60.0 T0:0.0 /0.0 T1:99.9 /295.0
Recv: B:38.6 /60.0 T0:0.0 /0.0 T1:101.6 /295.0
Recv: B:38.9 /60.0 T0:0.0 /0.0 T1:103.8 /295.0
Recv: B:39.2 /60.0 T0:0.0 /0.0 T1:105.6 /295.0
Recv: B:39.4 /60.0 T0:0.0 /0.0 T1:107.2 /295.0
Recv: B:39.7 /60.0 T0:0.0 /0.0 T1:109.5 /295.0
Recv: B:39.9 /60.0 T0:0.0 /0.0 T1:111.2 /295.0
Recv: B:40.1 /60.0 T0:0.0 /0.0 T1:112.9 /295.0
Recv: B:40.4 /60.0 T0:0.0 /0.0 T1:115.2 /295.0
Recv: B:40.7 /60.0 T0:0.0 /0.0 T1:116.8 /295.0
Recv: B:40.9 /60.0 T0:0.0 /0.0 T1:118.4 /295.0
Recv: B:41.2 /60.0 T0:0.0 /0.0 T1:120.6 /295.0
Recv: B:41.4 /60.0 T0:0.0 /0.0 T1:122.2 /295.0
Recv: B:41.6 /60.0 T0:0.0 /0.0 T1:123.6 /295.0
Recv: B:41.9 /60.0 T0:0.0 /0.0 T1:125.8 /295.0
Recv: B:42.1 /60.0 T0:0.0 /0.0 T1:127.4 /295.0
Recv: B:42.3 /60.0 T0:0.0 /0.0 T1:128.9 /295.0
Recv: B:42.6 /60.0 T0:0.0 /0.0 T1:131.0 /295.0
Recv: B:42.8 /60.0 T0:0.0 /0.0 T1:132.5 /295.0
Recv: B:43.0 /60.0 T0:0.0 /0.0 T1:134.1 /295.0
Recv: B:43.3 /60.0 T0:0.0 /0.0 T1:136.1 /295.0
Recv: B:43.5 /60.0 T0:0.0 /0.0 T1:137.6 /295.0
Recv: B:43.7 /60.0 T0:0.0 /0.0 T1:139.2 /295.0
Recv: B:44.0 /60.0 T0:0.0 /0.0 T1:141.1 /295.0
Recv: B:44.2 /60.0 T0:0.0 /0.0 T1:142.7 /295.0
Recv: B:44.4 /60.0 T0:0.0 /0.0 T1:144.2 /295.0
Recv: B:44.7 /60.0 T0:0.0 /0.0 T1:146.2 /295.0
Recv: B:44.9 /60.0 T0:0.0 /0.0 T1:147.7 /295.0
Recv: B:45.1 /60.0 T0:0.0 /0.0 T1:149.1 /295.0
Recv: B:45.4 /60.0 T0:0.0 /0.0 T1:151.1 /295.0
Recv: B:45.6 /60.0 T0:0.0 /0.0 T1:152.5 /295.0
Recv: B:45.8 /60.0 T0:0.0 /0.0 T1:153.9 /295.0
Recv: B:46.1 /60.0 T0:0.0 /0.0 T1:155.7 /295.0
Recv: B:46.3 /60.0 T0:0.0 /0.0 T1:157.1 /295.0
Recv: B:46.5 /60.0 T0:0.0 /0.0 T1:158.5 /295.0
Recv: B:46.8 /60.0 T0:0.0 /0.0 T1:160.4 /295.0
Recv: B:47.0 /60.0 T0:0.0 /0.0 T1:161.8 /295.0
Recv: B:47.2 /60.0 T0:0.0 /0.0 T1:163.7 /295.0
Recv: B:47.5 /60.0 T0:0.0 /0.0 T1:165.0 /295.0
Recv: B:47.7 /60.0 T0:0.0 /0.0 T1:166.3 /295.0
Recv: B:47.9 /60.0 T0:0.0 /0.0 T1:168.0 /295.0
Recv: B:48.2 /60.0 T0:0.0 /0.0 T1:169.4 /295.0
Recv: B:48.4 /60.0 T0:0.0 /0.0 T1:170.7 /295.0
Recv: B:48.6 /60.0 T0:0.0 /0.0 T1:172.5 /295.0
[... 209 lines cut ...]
Recv: ok
Send: N41 G1 X9.825 Y-9.825 E5.73358*117
Recv: ok
Send: N42 M105*17
Recv: ok B:59.2 /60.0 T0:0.0 /0.0 T1:279.9 /295.0
Send: N43 G1 X9.825 Y9.825 E6.05700*81
Recv: ok
Send: N44 G0 F9000 X9.103 Y9.351*114
Recv: ok
Send: N45 G1 F1200 X-9.351 Y-9.103 E6.48655*63
Recv: // Klipper state: Shutdown
Recv: ok
Send: N46 G0 F9000 X-9.103 Y-9.351*112
Recv: // Heater extruder1 not heating at expected rate
Recv: // See the 'verify_heater' section in config/example-extras.cfg
Recv: // for the parameters that control this check.
Recv: //
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! Heater extruder1 not heating at expected rate
Changing monitoring state from "Printing" to "Error: Heater extruder1 not heating at expected rate"
Changing monitoring state from "Error: Heater extruder1 not heating at expected rate" to "Offline (Error: Heater extruder1 not heating at expected rate)"
Connection closed, closing down monitor

KevinOConnor, 2019-05-23:
You'll need to attach the Klipper log file from the failed run in order for us to assist.

-Kevin

Geramy, 2019-05-23:
[klippylog.log](https://github.com/KevinOConnor/klipper/files/3210066/klippylog.log)
Attached is the log file, I can't even do a pid calibration

Geramy, 2019-05-23:
Kevin I ended up being able to do a calibration at 200c so I think i'll just leave it like this for now and try to do another calibration later at a higher temp. But so far its very very stable with the 200c temp calibration.

KevinOConnor, 2019-06-02:
I'm going to close this issue as it looks like the original issue was fixed (commit 6268c702).

-Kevin

closed 2019-06-02 by KevinOConnor