Klipper3d/klipper #1666: Unknown BLTouch pin numbers for sensor and control pins

adwsail, 2019-05-22:
I have an atmega2560 with a ramps 1.6 board running octopi very well. I am trying to add a BLTouch but have been totally unable to search out the correct "AR" pin numbers for the sensor and control pins. After several hours of searching and pulling my hair out I am forced to ask for help. I realize this is a sorry question to take up your time but I am truly at my wits end. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/3208973/klippy.log)

zwnk, 2019-05-22:
should be:
```
[bltouch]
sensor_pin: ^ar18
control_pin: ar11
```

adwsail, 2019-05-22:
You are a life saver. Thank you.

jakep82, 2019-05-22:
For future reference converting pin numbers is explained in the [FAQ](https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-convert-a-marlin-pin-number-to-a-klipper-pin-name).

adwsail, 2019-05-22:
Ive read the FAQ top to bottom, repeatedly. There is no reference to an actual number for the servo pins. I understad PA 11 would be ar11 for the the printer.cfg file but nowhere in any reference I found is there a pin number reference for the sensor_pin or control_pin. ZWNK actually had the correct answer to my question. But thank you for commenting.

jakep82, 2019-05-22:
The first result when searching for `RAMPS 1.6` is a link to [this RepRap page](https://reprap.org/wiki/RAMPS_1.6) which shows the servo pin numbers.  The link I previously provided explains how to convert any of those numbers to something that will work in Klipper.

adwsail, 2019-05-23:
I see now. Thank you.
Has anyone been able to get a real BLTouch working on a ramps 1.6/atmega2560/octopi platform? Perhaps even a working printer.cfg file? Pin_ and -down works fine, power on sel-test works fine, but the next test step, sending G28, never extends the probe, one needs to be quick with the M112 command to avoid a crash. This should probably be a new issue post, sorry.

KevinOConnor, 2019-06-02:
Was a solution found for this issue?  The last issue sounds like one of the BLTouch v3 issues that the (recently updated) documentation describes.  https://www.klipper3d.org/BLTouch.html

-Kevin

closed 2019-06-18 by KevinOConnor

OCD3DP, 2020-07-23:
I'm trying to figure out the sensor pin for my BLTouch using a Replicape. Does anyone have a working example of the config settings for this combo? Thanks in advance!

BlackStump, 2020-07-23:
I have never used a BLTouch with Replicape and I no longer have one to test but it depend on what pins you intend to connect your BLtouch to.
ie 
which servo
which endstop
[Replicape config](https://github.com/KevinOConnor/klipper/blob/master/config/generic-replicape.cfg#L130)
if you use endstop x2 servo_0  control pin number would be replicape:servo0
if you used endstop z2 as sensor pin it would be P9_18

my best guess as a example would be
>[bltouch]
sensor_pin: P9_18
control_pin: replicape:servo0

[image]

[Replicape_Rev_B3_pin-mux.pdf](https://github.com/KevinOConnor/klipper/files/4963980/Replicape_Rev_B3_pin-mux.pdf)