Klipper3d/klipper #1667: Delta printer Move out of range: error

richgain, 2019-05-22:
Hi
I've started using Klipper very successfully with my RichRap 3DR delta printer.
My problem is that I can't easily tell how large I can go in X and Y dimensions.
My bed is 250 mm diameter but I seem to be limited to about a 60 mm radius - not sure why.
The attached log file contains the following error:
  Move out of range: 53.733 34.690 0.200 [43.065]

According to the config file, delta_radius = 116.615116 and I can't seem to get anywhere near that.  I'm using radius = 60 for the [delta_calibrate] because if I go any higher it stops with an error.
Any suggestions?

[klippy.log](https://github.com/KevinOConnor/klipper/files/3209118/klippy.log)

BlackStump, 2019-05-22:
recheck your delta arm length, better still precisely measure the hinge to hinge length of your delta arm length.
arm_length = 181.196224 ~~~ is too small for a 250mm bed,
A mini kossel with a 170mm bed has 215mm arms
I have a 280mm bed with 372mm arms do a google on delta arm lengths vs horizontals frame length.

richgain, 2019-05-23:
Thanks. Those numbers were auto-generated by the delta_calibrate routine. The previous values were all 180 mm which is the length that I made them to be. I now realise that this is too short for the size of the frame and I will make some longer ones.
But my question is still valid. How do I find the diameter of the printable circle with the current hardware? I would like to know what scaling factor settings to use in my slicer without having the prints fail due to a 'Move out of range' error.

BlackStump, 2019-05-23:
I would suggest doing some research on Delta Printers and in particular to 
frame size
delta arm lengths
delta radius
and print radius
your physical hardware determines your print radius and the correct config to match those hardware details.
a starting point 
https://reprap.org/wiki/Delta_geometry
search for delta frame calculator which will usually give you what you seek.

richgain, 2019-05-23:
I was interested to know what 'Move out of range: 53.733 34.690 0.200 [43.065]' is telling me. 
Based on the assumption that the first three numbers represent X, Y and Z coordinates, I used the Pythagorean equation to calculate the square root of X square plus Y squared. The answer is almost exactly 64 mm, so until my longer arms arrive I think I shall assume that that is my printable radius.
Can anyone tell me what the fourth number means?

KevinOConnor, 2019-05-24:
The fourth number is the extruder position (43.065mm of filament extruded).

The location of the boundary checks are reported in the klipper log file on each startup.  For example, my printer reports:
```
Delta max build height 251.45mm (radius tapered above 209.49mm)
Delta max build radius 123.44mm (moves slowed past 110.49mm and 120.69mm)
```

-Kevin

richgain, 2019-05-24:
Perfect! Thanks, Kevin.

Here's mine:
```=======================
Delta max build height 326.87mm (radius tapered above 284.78mm)
Delta max build radius 63.05mm (moves slowed past 54.42mm and 61.22mm)

```

closed 2019-05-28 by richgain