Klipper3d/klipper #1669: Change of board TMC2130 sensorless homing no longer working

AxMod3DPrint, 2019-05-23:
I had tmc2130 sensorless homing working on a RAMPS Board but swapped out to MKS Gen L. Now sensorless homing is no longer working. Wiring is correct, pins have been triple checked, M119 says all pins open. Whatever SGT is set it is just crashing into the home position. log SGT differs due to testing.

Not sure what's going on here.

Log:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3213058/klippy.log)

TMC Dump:
```
Send: dump_tmc stepper=stepper_x
Recv: // ========== Write-only registers ==========
Recv: // IHOLD_IRUN: 00081616 IHOLD=22 IRUN=22 IHOLDDELAY=8
Recv: // TPWMTHRS:   00000000
Recv: // TPOWERDOWN: 00000000
Recv: // PWMCONF:    00050480 PWM_AMPL=128 PWM_GRAD=4 pwm_freq=1 pwm_autoscale=1
Recv: // COOLCONF:   000a0000 sgt=10
Recv: // ========== Queried registers ==========
Recv: // GCONF:      00000000
Recv: // GSTAT:      00000000
Recv: // IOIN:       110000f8 DCIN_CFG5=1 DRV_ENN_CFG6=1 DCO=1 VERSION=0x11
Recv: // TSTEP:      000fffff TSTEP=1048575
Recv: // XDIRECT:    00000000
Recv: // MSCNT:      00000004 MSCNT=4
Recv: // MSCURACT:   00f70006 CUR_A=6 CUR_B=247
Recv: // CHOPCONF:   13028384 toff=4 hend=7 TBL=1 vsense=1 MRES=3(32usteps) intpol=1
Recv: // DRV_STATUS: 80160000 CS_ACTUAL=22 stst=1
Recv: // PWM_SCALE:  00000000
Recv: // LOST_STEPS: 00000000
```

AxMod3DPrint, 2019-05-24:
Update: there is a 10mv voltage spike when measured when homing, so the drivers are OK, when pin is closed to ground using the probe, it registers as triggered, so the board itself is OK, Changed the pins, still the same result. Does anyone have any ideas?

KevinOConnor, 2019-05-25:
Looks like you've wired the diag1 pin to an endstop pin on the board.  A number of these "all in one ramps" boards put capacitors and pullups on the endstop pins which may be messing with the detection.  You might want to try with a pin not associated with an endstop.

Otherwise, unfortunately, I don't have any insight.

-Kevin

AxMod3DPrint, 2019-05-25:
I've tried ar2 on X with the same result, but will give another different pin, say off the servo header and see how it goes. I may have to switch back to the RAMPS and use software serial.

AxMod3DPrint, 2019-05-27:
Servo header ar3 used with the same result. I'm none too sure what's up here.

AxMod3DPrint, 2019-06-02:
Just tried Marlin and I'm getting an overtemp on E and can't clear it, so using that avenue is a bust..  not sure if that would help to the issue..

AxMod3DPrint, 2019-06-06:
Here's a kicker... just changed back to the RAMPS that I was using and I'm getting the same thing, so something somewhere is screwy....
Log:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3262727/klippy.log)

jakep82, 2019-06-06:
Sensorless homing is working fine for me. I updated Klipper within the last couple weeks, so I don't think this is a software issue.

jakep82, 2019-06-06:
Have you tried using negative SGT values?  I switched to 0.9 degree motors and I ended with -12 versus 4 for my old 1.8 degree motors.

AxMod3DPrint, 2019-06-06:
I'm really not sure what it is as If I trigger the endstop by removing the ! and restarting, it shows up as triggered. I've tried negative and positive SGT values and they are being written to the driver. There is a voltage spike when the SGT triggers, so it _should_ work but it's not. I'm stumped at this point. I may have to put standard endstops back on..

KevinOConnor, 2019-06-28:
Any further updates on this?  Unfortunately, I don't have much advice to offer.

-Kevin

AxMod3DPrint, 2019-07-19:
Apologies, just getting back to this now and still no change. I'm very close to saying sod it and putting standard end stops back on.

KevinOConnor, 2019-09-21:
Okay, I'm going to close this issue.  It does not appear to be a systemic issue with Klipper, and it doesn't look like anyone has further advice to offer.

-Kevin

closed 2019-09-21 by KevinOConnor