Klipper3d/klipper #1670: how to run bltouch.py

adwsail, 2019-05-23:
Is there a way to run bltouch.py in the octoprint terminal?

Paulbsh, 2019-05-23:
Why would you want to do that?

adwsail, 2019-05-25:
Curiosity.

Paulbsh, 2019-05-25:
Not sure if this is a valid post for the issues section.
There is no sense in running this py script on its own

adwsail, 2019-05-26:
With all due respect my question was "can you run a python script from the
octoprint terminal window?" So far no one has answered a simple yes or no
question. Why I want to do so is my business and has absolutely nothing to
do with question. Klipper and Octoprint is a great pairing and I am
exploring various aspects of its programming and the limits of its current
capabilities. If you know the answer to my question one way or the other,
great. Since I am not asking about a specific 'issue' with either program I
fail to understand the questions re motivation for doing so. Isn't the fact
that I am asking a question enough of a 'reason'?

Paulbsh, 2019-05-26:
With all respect from my side, this section is meant for issues of the Klipper software, not for question out of curious reasons.
As far as I know, the Octo terminal is not able to run dedicated python scripts.
You would have to do it on the Raspi itself.

adwsail, 2019-05-26:
Thank you, since programming is what I do that helps.

KevinOConnor, 2019-05-29:
I don't understand the original question.  If an issue was encountered with the bltouch then a log file from the session will be needed.  If one is looking to program with Klipper then start with the documentation at: https://www.klipper3d.org/Overview.html#developer-documentation

-Kevin

closed 2019-05-30 by klipper-gitissuebot