Klipper3d/klipper #1671: No step/dir pulses generated on sam3x8e (using delta kinematics)

simonkuehling, 2019-05-23:
Hi,

i am setting up Klipper on a UDOO Quad board which has an Arduino Due compatible sam3x8e onboard. Got everything up and running so far, but the MCU does not seem to produce any _step_ or _direction_ pulses on its pins. I double-checked pin numbers and measured step+dir pins on all three steppers on an oscilloscope. No pulses whatsoever.
The _enable_ pins are properly working, though. Also the output_pin configuration for switching the power supply is working.

So far I have tested via STEPPER_BUZZ as well as FORCE_MOVE commands for all three steppers without success.

Does the log tell anything that I am missing?
How would I go forward to track this down further? 

[klippy.log](https://github.com/KevinOConnor/klipper/files/3213796/klippy.log)

KevinOConnor, 2019-05-25:
That's certainly odd.  It's a popular board, and to the best of my knowledge the code works as intended.

-Kevin

simonkuehling, 2019-05-27:
Hmm.. 
In order to isolate if this issue may be MCU code related or not, I moved all three steppers to a separate, different MCU for testing. It is an Adafruit Metro Mini (ATMega328P) connected via USB.
Surprisingly enough, it is still the exact same bahavior: `enable` pins are working, `step` and `direction` are inactive no matter what.

Is it possible that these actions are not scheduled by the host for some reason?

Here is a log file of this recent test:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3224031/klippy.log)

simonkuehling, 2019-05-28:
@KevinOConnor, I have interesting results from further testing - this is what I did:

Part One:
* Attach an Arduino Uno (ATMega328P) via USB on the UDOO
* Flash and configure this microcontroller as the main klipper MCU
* Check step/dir pulse generation via "STEPPER_BUZZ" command and Oscilloscope

Part Two:
* Set up Klipper on my Ubuntu 18.04 system on a ThinkPad
* use the exact same printer.cfg from part one
* Take the Arduino over to my ThinkPad (without re-flashing!), run Klipper
* Check step/dir pulse generation via "STEPPER_BUZZ" command and Oscilloscope

Results:
On the UDOO I could not get any pulses (like before). After carrying the MCU over to my ThinkPad, it worked flawlessly!

I would conclude that there is something wrong on the klippy side of the UDOO setup - looks like the step generation is somehow not running on the UDOO. Do you have a hint what I could check for?

Here are both log files for direct comparison:
[klippy-UDOO.log](https://github.com/KevinOConnor/klipper/files/3226699/klippy-UDOO.log)
[klippy-THINKPAD.log](https://github.com/KevinOConnor/klipper/files/3226700/klippy-THINKPAD.log)

KevinOConnor, 2019-05-29:
If the micro-controller is powered via USB, it could be something with the power supply of the "UDOO".

It seems unlikely it's the host code.  You could issue an M112 after the STEPPER_BUZZ on both machines and compare the commands sent to the micro-controller.

-Kevin

simonkuehling, 2019-05-31:
Thanks for the hint - I just issued a single FORCE_MOVE command followed by a M112 on both systems. Interestingly enough, it seems that the steps are actually not even transmitted to the MCU on the UDOO setup. Like before, I used the same Arduino board without reflashing on both systems.

Snippets from the logs:
THINKPAD:
```
Sent 82 899753.673415 899753.673175 6: seq: 1b, get_clock
Sent 83 899754.658229 899754.657989 6: seq: 1c, get_clock
Sent 84 899755.643115 899755.642875 6: seq: 1d, get_clock
Sent 85 899756.321042 899756.320042 25: seq: 1e, set_next_step_dir oid=0 dir=1, queue_step oid=0 interval=223569730 count=1 add=0, schedule_digital_out oid=3 clock=221946685 value=0
Sent 86 899756.425159 899756.424679 12: seq: 1f, queue_step oid=0 interval=40000 count=1 add=0
Sent 87 899756.627470 899756.627230 6: seq: 10, get_clock
Sent 88 899757.612437 899757.612197 6: seq: 11, get_clock
Sent 89 899758.597534 899758.597294 6: seq: 12, get_clock
```
UDOO:
```
Sent 86 1381628.225036 1381628.224796 6: seq: 15, get_clock
Sent 87 1381629.210189 1381629.209949 6: seq: 16, get_clock
Sent 88 1381630.195314 1381630.195074 6: seq: 17, get_clock
Sent 89 1381630.561631 1381630.561151 12: seq: 18, schedule_digital_out oid=3 clock=75912932 value=0
Sent 90 1381631.180367 1381631.180127 6: seq: 19, get_clock
Sent 91 1381632.164636 1381632.164396 6: seq: 1a, get_clock
Sent 92 1381633.149723 1381633.149483 6: seq: 1b, get_clock
```

Full logs:
[klippy-THINKPAD.log](https://github.com/KevinOConnor/klipper/files/3241212/klippy.log)
[klippy-UDOO.log](https://github.com/KevinOConnor/klipper/files/3241215/klippy.log)

Could it be an issue with the older gcc version used to compile the c_helper code? I already manually run the compilation to look for any errors or warnings, but that is finishing without complaints...

KevinOConnor, 2019-05-31:
Well, that's certainly odd.  Try removing klippy/chelper/*.so and then restart klippy (sudo service klipper restart) to see if a recompile fixes it.  If that doesn't fix it, you could try messing with the COMPILE_CMD parameters at the top of `klippy/chelper/__init__.py` to see if it that's the cause of the problem (in particular, try removing the " -flto -fwhole-program -fno-use-linker-plugin" line).  Be sure to remove the old library (`rm  klippy/chelper/*.so`) and do a full restart (sudo service klipper restart) on any software change.  We're not really doing anything particularly special in the host C code, so I'm surprised it would be an issue.

-Kevin

KevinOConnor, 2019-05-31:
Also, what does `gcc --version ; ld --version` report on your machines?

-Kevin

simonkuehling, 2019-05-31:
I already tried forcing a recompile by removing the .so without success so far, but I'll fiddle with the build arguments to check if this makes any difference - will keep you posted about the results...

gcc and ld versions are:

UDOO
```
$ gcc --version
gcc (Ubuntu/Linaro 4.8.4-2ubuntu1~14.04.4) 4.8.4
$ ld --version
GNU ld (GNU Binutils for Ubuntu) 2.24
```
THINKPAD
```
$ gcc --version
gcc (Ubuntu 7.4.0-1ubuntu1~18.04) 7.4.0
$ ld --version
GNU ld (GNU Binutils for Ubuntu) 2.30
```

simonkuehling, 2019-06-03:
I forced a recompile without the ```-flto -fwhole-program -fno-use-linker-plugin``` parameters today - but the results are the same, no stepper commands transmitted...

[klippy-NEW-CHELPER.log](https://github.com/KevinOConnor/klipper/files/3246737/klippy-NEW-CHELPER.log)

Do you have a hint where I would optimally place some logging output to watch the data structures passed into the C code? Maybe a comparison further up the code flow can reveal where exactly things go south...

KevinOConnor, 2019-06-03:
I'd sprinkle `logging.debug()` calls throughout the code flow described at https://www.klipper3d.org/Code_Overview.html#code-flow-of-a-move-command .  For the C code, you can sprinkle `errorf()` calls.

-Kevin

simonkuehling, 2019-06-04:
Thanks, the ```errorf()``` is very helpful here!

So far I could find out that code flow seems to abruptly end in the itersolve_gen_steps() function. As suggested I've thrown some output lines in (code below). It never goes past the ```#3``` statement just before the goto label (log excerpt below). 

```
// Generate step times for a stepper during a move
int32_t __visible
itersolve_gen_steps(struct stepper_kinematics *sk, struct move *m)
{
errorf("in itersolve.c --> itersolve_gen_steps()");
    struct stepcompress *sc = sk->sc;
    sk_callback calc_position = sk->calc_position;
    double half_step = .5 * sk->step_dist;
    double mcu_freq = stepcompress_get_mcu_freq(sc);
    struct timepos last = { 0., sk->commanded_pos }, low = last, high = last;
    double seek_time_delta = 0.000100;
    int sdir = stepcompress_get_step_dir(sc);
    struct queue_append qa = queue_append_start(sc, m->print_time, .5);
    for (;;) {
errorf("itersolve_gen_steps --> #1");
        // Determine if next step is in forward or reverse direction
        double dist = high.position - last.position;
        if (fabs(dist) < half_step) {
errorf("itersolve_gen_steps --> #3");
        seek_new_high_range: 
errorf("itersolve_gen_steps --> #4");
            if (high.time >= m->move_t)
errorf("itersolve_gen_steps --> #5");
                // At end of move
                break;
            // Need to increase next step search range
errorf("itersolve_gen_steps --> #6");
            low = high;
            high.time = last.time + seek_time_delta;
            seek_time_delta += seek_time_delta;
            if (high.time > m->move_t)
                high.time = m->move_t;
            high.position = calc_position(sk, m, high.time);
errorf("itersolve_gen_steps --> #7");
            continue;
        }
```
from the log:
```
FORCE_MOVE stepper_a distance=0.100 velocity=20.000 accel=0.000
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
in itersolve.c --> move_fill()
in itersolve.c --> move_fill()
 *** FILLING MOVE *** print_time 11.4281645 accel_t 0.0 cruise_t 0.005 accel_t 0.0 dist 0.1 cruise_v 20.0 accel 0.0
in itersolve.c --> itersolve_gen_steps()
itersolve_gen_steps --> #1
itersolve_gen_steps --> #3
itersolve_gen_steps --> #4
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
in stepcompress.c --> step_compress_flush()
```

simonkuehling, 2019-06-05:
Seems like I am making progress at last...

In itersolve.c --> itersolve_gen_steps() I can see that on the UDOO, the call to ```calc_position()``` always returns ```0```and consequently queue_append() is never called, while on the THINKPAD it is delivering correct values and steps are queued.
@KevinOConnor, I am not quite sure if I followed the code correctly - is ```calc_position()``` invoking ```kin_delta.c --> delta_stepper_calc_position()``` here?

This is my code for ```itersolve_gen_steps()```:
```
// Generate step times for a stepper during a move
int32_t __visible
itersolve_gen_steps(struct stepper_kinematics *sk, struct move *m)
{
    struct stepcompress *sc = sk->sc;
    sk_callback calc_position = sk->calc_position;
    double half_step = .5 * sk->step_dist;
    double mcu_freq = stepcompress_get_mcu_freq(sc);
    struct timepos last = { 0., sk->commanded_pos }, low = last, high = last;
    double seek_time_delta = 0.000100;
    int sdir = stepcompress_get_step_dir(sc);
    struct queue_append qa = queue_append_start(sc, m->print_time, .5);
    for (;;) {
        // Determine if next step is in forward or reverse direction
        double dist = high.position - last.position;
errorf("\t itersolve_gen_steps --> high.position: %f \t last_position: %f \t dist: %f", high.position, last.position, dist);
        if (fabs(dist) < half_step) {
        seek_new_high_range:
            if (high.time >= m->move_t)
                // At end of move
                break;
            // Need to increase next step search range
            low = high;
            high.time = last.time + seek_time_delta;
            seek_time_delta += seek_time_delta;
            if (high.time > m->move_t)
                high.time = m->move_t;
            high.position = calc_position(sk, m, high.time);
            continue;
        }
        int next_sdir = dist > 0.;
        if (unlikely(next_sdir != sdir)) {
            // Direction change
            if (fabs(dist) < half_step + .000000001)
                // Only change direction if going past midway point
                goto seek_new_high_range;
            if (last.time >= low.time && high.time > last.time) {
                // Must seek new low range to avoid re-finding previous time
                high.time = (last.time + high.time) * .5;
                high.position = calc_position(sk, m, high.time);
                continue;
            }
            int ret = queue_append_set_next_step_dir(&qa, next_sdir);
            if (ret)
                return ret;
            sdir = next_sdir;
        }
        // Find step
        double target = last.position + (sdir ? half_step : -half_step);
        struct timepos next = itersolve_find_step(sk, m, low, high, target);
        // Add step at given time
errorf("\t itersolve_gen_steps --> adding step to queue");
        int ret = queue_append(&qa, next.time * mcu_freq);
        if (ret)
            return ret;
        seek_time_delta = next.time - last.time;
        if (seek_time_delta < .000000001)
            seek_time_delta = .000000001;
        last.position = target + (sdir ? half_step : -half_step);
        last.time = next.time;
        low = next;
        if (last.time >= high.time)
            // The high range is no longer valid - recalculate it
[... 18 lines cut ...]
	 itersolve_gen_steps --> high.position: 0.000000 	 last_position: 0.000000 	 dist: 0.000000
```
log excerpt from THINKPAD:
```
FORCE_MOVE stepper_a distance=0.200 velocity=20.000 accel=0.000
	 itersolve_gen_steps --> high.position: 0.000000 	 last_position: 0.000000 	 dist: 0.000000
	 itersolve_gen_steps --> high.position: 0.002000 	 last_position: 0.000000 	 dist: 0.002000
	 itersolve_gen_steps --> high.position: 0.004000 	 last_position: 0.000000 	 dist: 0.004000
	 itersolve_gen_steps --> high.position: 0.008000 	 last_position: 0.000000 	 dist: 0.008000
	 itersolve_gen_steps --> high.position: 0.016000 	 last_position: 0.000000 	 dist: 0.016000
	 itersolve_gen_steps --> high.position: 0.032000 	 last_position: 0.000000 	 dist: 0.032000
	 itersolve_gen_steps --> adding step to queue
	 itersolve_gen_steps --> high.position: 0.032000 	 last_position: 0.050000 	 dist: -0.018000
	 itersolve_gen_steps --> high.position: 0.050000 	 last_position: 0.050000 	 dist: 0.000000
	 itersolve_gen_steps --> high.position: 0.075000 	 last_position: 0.050000 	 dist: 0.025000
	 itersolve_gen_steps --> high.position: 0.125000 	 last_position: 0.050000 	 dist: 0.075000
	 itersolve_gen_steps --> adding step to queue
	 itersolve_gen_steps --> high.position: 0.125000 	 last_position: 0.100000 	 dist: 0.025000
	 itersolve_gen_steps --> high.position: 0.125000 	 last_position: 0.100000 	 dist: 0.025000
	 itersolve_gen_steps --> high.position: 0.175000 	 last_position: 0.100000 	 dist: 0.075000
	 itersolve_gen_steps --> adding step to queue
	 itersolve_gen_steps --> high.position: 0.175000 	 last_position: 0.150000 	 dist: 0.025000
	 itersolve_gen_steps --> adding step to queue
	 itersolve_gen_steps --> high.position: 0.200000 	 last_position: 0.200000 	 dist: 0.000000
```

KevinOConnor, 2019-06-06:
> I am not quite sure if I followed the code correctly - is calc_position() invoking kin_delta.c --> delta_stepper_calc_position() here?

Yes - `calc_position()` should be a function pointer to the kinematics code (eg, `delta_stepper_calc_position()`) for the given stepper motor.

-Kevin

simonkuehling, 2019-06-06:
Ok, thanks for the verification. I just understood that FORCE_MOVE is using the function from kin_cartesian.c and not the delta kinematics... which makes sense of course...

I finally spotted where exactly the data gets corrupted on the UDOO: It is where the C routine ```itersolve.c --> move_fill()``` is called from the python code in ```force_move.py --> manual_move()```. The parameters that are arriving in ```move_fill()``` are not identical to what is put in from the python side - and looks a bit strange when you look at the huge random```accel```value.
I have no idea what is going on here - do you?

debugging lines in ```force_move.py --> manual_move()```
```
def manual_move(self, stepper, dist, speed, accel=0.):
        toolhead = self.printer.lookup_object('toolhead')
        print_time = toolhead.get_last_move_time()
        prev_sk = stepper.set_stepper_kinematics(self.stepper_kinematics)
        stepper.set_position((0., 0., 0.))
        accel_t, cruise_t, cruise_v = calc_move_time(dist, speed, accel)
        logging.info("force_move.py --> manual_move()")
        logging.info(" \t\t *** print_time: %f", print_time)
        logging.info(" \t\t *** accel_t: %f", accel_t)
        logging.info(" \t\t *** cruise_t: %f", cruise_t)
        logging.info(" \t\t *** dist: %f", dist)
        logging.info(" \t\t *** cruise_v: %f", cruise_v)
        logging.info(" \t\t *** accel: %f", accel)
        self.move_fill(self.cmove, print_time, accel_t, cruise_t, accel_t,
                       0., 0., 0., dist, 0., 0., 0., cruise_v, accel)
        stepper.step_itersolve(self.cmove)
        stepper.set_stepper_kinematics(prev_sk)
        toolhead.dwell(accel_t + cruise_t + accel_t)
```
debugging lines in ```itersolve.c --> move_fill()```
```
// Populate a 'struct move' with a velocity trapezoid
void __visible
move_fill(struct move *m, double print_time
          , double accel_t, double cruise_t, double decel_t
          , double start_pos_x, double start_pos_y, double start_pos_z
          , double axes_d_x, double axes_d_y, double axes_d_z
          , double start_v, double cruise_v, double accel)
{
errorf("in itersolve.c --> move_fill()");
errorf("\t\t *** print_time: %f", print_time);
errorf("\t\t *** accel_t: %f", accel_t);
errorf("\t\t *** cruise_t: %f", cruise_t);
errorf("\t\t *** decel_t: %f", decel_t);
errorf("\t\t *** start_pos_x: %f", start_pos_x);
errorf("\t\t *** start_pos_y: %f", start_pos_y);
errorf("\t\t *** start_pos_z: %f", start_pos_z);
errorf("\t\t *** axes_d_x: %f", axes_d_x);
errorf("\t\t *** axes_d_y: %f", axes_d_y);
errorf("\t\t *** axes_d_z: %f", axes_d_z);
errorf("\t\t *** start_v: %f", start_v);
errorf("\t\t *** cruise_v: %f", cruise_v);
errorf("\t\t *** accel: %f", accel);

    // Setup velocity trapezoid
    m->print_time = print_time;
    m->move_t = accel_t + cruise_t + decel_t;
    m->accel_t = accel_t;
    m->cruise_t = cruise_t;
    m->cruise_start_d = accel_t * .5 * (cruise_v + start_v);
    m->decel_start_d = m->cruise_start_d + cruise_t * cruise_v;

    // Setup for accel/cruise/decel phases
    m->cruise_v = cruise_v;
    m->accel.c1 = start_v;
    m->accel.c2 = .5 * accel;
    m->decel.c1 = cruise_v;
    m->decel.c2 = -m->accel.c2;

    // Setup for move_get_coord()
    m->start_pos.x = start_pos_x;
    m->start_pos.y = start_pos_y;
    m->start_pos.z = start_pos_z;
[... 65 lines cut ...]
 		 *** print_time: 8.103451
 		 *** accel_t: 0.000000
 		 *** cruise_t: 0.010000
 		 *** dist: 0.200000
 		 *** cruise_v: 20.000000
 		 *** accel: 0.000000
in itersolve.c --> move_fill()
		 *** print_time: 8.103451
		 *** accel_t: 0.000000
		 *** cruise_t: 0.010000
		 *** decel_t: 0.000000
		 *** start_pos_x: 0.000000
		 *** start_pos_y: 0.000000
		 *** start_pos_z: 0.000000
		 *** axes_d_x: 0.200000
		 *** axes_d_y: 0.000000
		 *** axes_d_z: 0.000000
		 *** start_v: 0.000000
		 *** cruise_v: 20.000000
		 *** accel: 0.000000
	 itersolve_gen_steps --> adding step to queue
	 itersolve_gen_steps --> adding step to queue
	 itersolve_gen_steps --> adding step to queue
	 itersolve_gen_steps --> adding step to queue
```

KevinOConnor, 2019-06-06:
As a wild guess, there might be a problem with libffi on the target machine.  If you try the branch in PR #1465 - does that improve results?

-Kevin

simonkuehling, 2019-06-07:
Yes! PR #1465 totally resolves this issue - all parameters passed into move_fill are arriving in good condition!

mkuf, 2019-11-25:
On a unrelated Note, Ubuntu 14.04 went EOL on 31/04/19. 
I'm running a self built 18.04 Image on my Udoo Quad and haven't experienced any of the described issues. 
@simonkuehling If you're interested, I'll lend you a hand in building an up to date Image.

simonkuehling, 2019-12-30:
@mkuf, sorry for the late reply and thanks for your offer. The UDOO system was so far only a "try to make it work in a pre-existing and slightly outdated setup" while avoiding to rebuild/upgrade the OS... The UDOO will be replaced by more recent hardware anyway soon.

-Simon

simonkuehling, 2021-08-23:
Although this "issue" is probably technically still unresolved  - as @mkuf pointed out, the target system went EOL quite a while ago now and I am closing this for now.

closed 2021-08-23 by simonkuehling