Klipper3d/klipper #1672: Heater /Transistor burnout

messyhippy, 2019-05-23:
Hi, Am in the process of upgrading my ender3 pro to run dual extruders. I have purchased an MKS Gen v1.4 Board, and followed all the instructions for setting up klipper. All seemed good, temp sensors working ok, steppers working ok. then I tried to run the PID calibrate on extruder 0. The temperature immediately climbed beyond 250c, at which point the extruder caught fire, as did the mks board. No Idea why, before I go and buy replacment parts, any ideas what might have happened??
Cheers
[klippy.log](https://github.com/KevinOConnor/klipper/files/3214320/klippy.log)

jakep82, 2019-05-23:
Did you power a 12V heater cartridge with 24V?

BlackStump, 2019-05-23:
sounds like a serious hardware or wiring problem or as Jake said a mismatch of parts, a multimeter is prerequisite for upgrading/replacing especially with heaters to check the resistance and verify the heater is for the correct voltage and wattage.
It is not a Klipper issue if you were wondering.

KevinOConnor, 2019-05-29:
From the log, it appears the Klipper software worked as intended - it detected a series of heating faults and each time it disabled the heater and went into an error state.  Best guess is that there was a hardware issue (as discussed above).

-Kevin

messyhippy, 2019-06-04:
Thanks for the feedback, I didnt really think it was a klipper issue. Still a bit bemused, the heaters where correct, wiring all looked good (except the fried bits!). Ive got klipper working with my original ender 3 board no problems at all. Back to the drawing board for the dual extruder upgrade for now.

closed 2019-06-04 by messyhippy