Klipper3d/klipper #1673: Internal error on command: "G28" Adafruit Grand Central (samd51p20a) with RADDS v1.6

Oranginator, 2019-05-24:
Hi together,

I want to use Klipper on an Adafruit Grand Central M4 Express featuring the SAMD51 together with RADDS v1.6 and TMC5160.
When I try to home with the G28 command I get the error: Internal error on command:"G28"

I'm using soft-spi as I couldn't get my hardware SPI working... it's also hardwired like shown in the picture but I couldn't find the MCU-pins I had to use in the "Sercom"-Path of klipper.
[image]

Thanks in advance!

[klippy.log](https://github.com/KevinOConnor/klipper/files/3215447/klippy.log)

KevinOConnor, 2019-05-24:
The log seems to indicate the micro-controller code got corrupted.  It's hard to tell, however, as the log itself appears to be truncated.  (We need the full log to assist.)  I'd try reflashing the micro-controller (make clean ; make ; make flash) to see if that changes the result.  If you continue to get a failure, then please try attaching the full log again.

I'm not sure anyone has run Klipper on the Grand Central.  (We've run Klipper on a several different Adafruit M4 boards and other SAMD51 chips - but I'm not sure anyone has tried this particular board yet.)

The pins you have for spi in your config (PC24, PC25, PC26) are not valid sercom pins.  (So, hardware spi can not be used with them.)  What pins did you hook them up to on the Grand Central?

-Kevin

Oranginator, 2019-05-26:
Hi Kevin,

I followed the steps you mentioned (make clean ; make ; make flash).

I now hooked them up on the ICSP pins 3x2 in the middle... to select the DUE equivalent pins I used the following maps. How can I verfiy which pins are physically used... maybe this is the main problem.

https://cdn-learn.adafruit.com/assets/assets/000/068/964/original/adafruit_products_Grand_Central_Sch2.png?1547072264&fbclid=IwAR1NlgHRd8vwtFUK5AoMVLzxsGCzbM56gG3ItUm9yS5dhLHvh-rBK1IgmR0

https://cdn-learn.adafruit.com/assets/assets/000/069/175/original/adafruit_products_Grand_Central_Sch1.png?1547234817&fbclid=IwAR1NlgHRd8vwtFUK5AoMVLzxsGCzbM56gG3ItUm9yS5dhLHvh-rBK1IgmR0

I also tried SAMD SERCOM configuration as descirbed in the example file with no sucess.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3220408/klippy.log)

KevinOConnor, 2019-05-29:
The log seems to indicate that the micro-controller state somehow got severely corrupted.  One thing you could try is to reduce the config to the most simplest case possible and see if that works.  Then build up a more complex config to try and determine what the source of the failure is.

Otherwise, it may require someone with the board and an arm chip debugger to figure out what is the source of the corruption.

It's also possible there is an electrical instability in the board/wiring that is corrupting the micro-controller state.

-Kevin

KevinOConnor, 2019-06-28:
Any further updates on this?

-Kevin

Oranginator, 2019-06-28:
I'm in vacation atm. Changed to Arduino Due this works for me. Thanks for asking.

closed 2019-07-21 by Oranginator

KevinOConnor, 2019-07-31:
FYI, it appears selecting the "internal clock" (instead of the "32Khz clock") in "make menuconfig" is a workaround for this issue.

-Kevin