Klipper3d/klipper #1674: Creality Ender-3 display on SKR 1.3 and other

povlhp, 2019-05-24:
Not really a bug - Should likely be a pull request.
Can somebody put this into the BigTreeTech SKR 1.3 config ? The Creality printers are so popular that it is important to support its display controller. Finding pin mappings on the Internet was impossible (They merged 2x 10-pin to 1x 10-pin)

```
## Creality Ender-3 / CR-10 Full Graphic Smart Controller - 1x 10 pin cable ##

[display]
lcd_type: st7920
cs_pin: P1.19
sclk_pin: P1.21
sid_pin: P1.23
encoder_pins: ^P1.18, ^P1.20
click_pin: !P0.28
#
[output_pin beeper]
pin: P1.30

```

And here is the  secret undiscovered PIN layout of their connector (The button pins might need a swap)

```
5V        GND
SID       ?
SCLK      Btn_En1
CS        Btn_En2
Click    Beeper
```

povlhp, 2019-05-24:
A log to make the bot happy
[klippy.zip](https://github.com/KevinOConnor/klipper/files/3216898/klippy.zip)

KevinOConnor, 2019-05-25:
Okay, thanks - commit 9e1d7950.

-Kevin

closed 2019-05-26 by povlhp