Klipper3d/klipper #1676: Anet A8 extruder calibration issue

XXX-ManZG, 2019-05-24:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3218478/klippy.log)

When trying to calibrate extruder, it extrudes only about 60mm instead of around 100mm with G1 E100 F100. Printer is actually a little over extruding, so that is wrong. There is no filament skipping.

adwsail, 2019-05-25:
When I got my used ante a8 it printed ok but in checking the extruder calibration it only put out 22mm instead of the expected 100. So I did the math and corrected the step size. The next attempts jammed in the extruder and I fought for 2 weeks to get it to print correctly. The flow rate is the only other variable that controls the amount of plastic pushed through directly. So now my extruder puts out what is expected but to avoid jams I must run it at 70% flow rate and 210 degrees c for matterhackers pla filament. It’s still underextruding and on the verge of jamming. 

So I’m sorry to say I have not found an explanation as to why much less how to fix it properly. The numbers don’t lie, trust the math. You can get it to print by deliberately setting your step size wrong, which it currently is, or set it right and reduce the flow rate. Good luck finding info on why it’s necessary.

XXX-ManZG, 2019-05-25:
My printer was perfect on Marlin, but on Klipper it doesn't work good. All it needs is calibration and I can't do it for some reason. I really like Klipper and would like to use it but if there is no solution I will have to go back to Marlin to use it.

adwsail, 2019-05-25:
You’ve gone from a Ford to a Porsche in some ways. Learning how to calibrate the extruder, x and y axis is very simple and requires only basic math skills. I would strongly urge you to take the time to learn how. The internet is full of calibration “how to’s”. The z axis is, in principle, the same but there are some potential gotchas, auto bed leveling systems for example. The huge plus to learning how is you will learn a great deal more about your machine and the fundamental laws of physics governing its operation. But if for whatever reason you don’t want to learn then by all means go back to Marlin. Changing the fundamental firmware on a printer is not for the faint of heart. It’s not hard but it’s more than simple load and go. Good luck.

XXX-ManZG, 2019-05-26:
I am learning and want to learn more, but the thing is it's not working as it should and I can't get any good advice.

adwsail, 2019-05-26:
Well I'm a newbie myself but I'm willing to help if I can. For starters
lets get that extruder calibrated properly. Are you a Youtube guy or a text
based guy? I prefer text directions myself but many people work better with
videos. Is your extruder a direct connect or bowden tube setup? For
calibrating the extruder it really doesn't matter but it will help in my
explanation of how to do the calibration.

XXX-ManZG, 2019-05-26:
Thank you for helping!
I can work with video or text, it's the same to me. 
My printer is stock Anet A8, so direct drive.
I have tried calibrating by using this (i believe very good) video: https://www.youtube.com/watch?v=eqCp1IiR3do

theopensourcerer, 2019-05-26:
Read this. https://mattshub.com/2017/04/19/extruder-calibration/

Then you just need to change steps/mm into mm/step by dividing by 1, i.e.

1 / (Non Klipper in steps/mm) = value for Klipper

So if you get a result of 172 steps/mm, for the Klipper configuration it would be:

1 / 172 = 0.00581

adwsail, 2019-05-26:
I just watched the video you used and yes, Brian is very good. I have a
bowden tube set up so I dont need to actually run any filament through the
hotend but the principle is the same, make a reference mark that will been
seen after running out "100mm", measure what actually came out, plug the
numbers in and do the math exactly as he says. So what specifically is your
question/issue? Trust the math. My default was 0.0105, my adjusted step
distance for klipper is 0.0024. Now when I issue a G1 E100 F100 command in
the terminal I get precisely 100mm.

XXX-ManZG, 2019-05-26:
My problem is (was) actually that it extrudes always the same no matter what I put in configuration.

But, the problem is solved!!!
I just formatted sd card and installed octoprint and klipper again and all is good after calibration. Must be some glitch somewhere.

Thank you for your help!

adwsail, 2019-05-26:
Anytime.

KevinOConnor, 2019-06-09:
I'm going to close this as it appears a solution was found for the original issue.

-Kevin

closed 2019-06-09 by KevinOConnor