Klipper3d/klipper #1678: Problem with new MKS Gen L V1.0 - usb-com - QinHeng Electronics HL-340

Nurmukhamed, 2019-05-25:
I have
Orange Pi PC 2
MKS Gen L V1.0
Armbian
Octoprint
Klipper.

after install i found that i cant extrude filament. then i replaced E0 to E1 - no luck. after i guess that my LV8729 driver is fault, so i replaced. no luck.
Then i replace firmware to Marlin, and anything works well.

i guess problem in usb-com because at old card it is works well.

DBa2016, 2019-05-28:
> I have
> Orange Pi PC 2
> MKS Gen L V1.0
> Armbian
> Octoprint
> Klipper.
>
> after install i found that i cant extrude filament. then i replaced E0 to E1 - no luck. after i guess that my LV8729 driver is fault, [...]

Okay, since we are at guesswork anyway (without a logfile, nothing else is possible)... I am assuming/guessing that only extrusion fails, normal movement and heating etc. works fine (otherwise you would have mentioned that, right?). Please try heating the hotend to a working temperature first (default is I think 170°C, the settings are "min_temp" and "max_temp" in the [extruder] section) and then retry moving the extruder. Observe the "terminal" tab for errors and let us know the outcome.

closed 2019-06-01 by klipper-gitissuebot

elcojacobs, 2019-06-01:
I'm using the MKS gen L without any problems. You do have to configure all your pins correctly of course.