Klipper3d/klipper #1679: [DEVELOPMENT] - MCU Interrupts

smark-, 2019-05-25:
Hi, I'd like to implement in Klipper a feature to make use of the TMC2130 overheating warning flag over DIAG0/DIAG1. I want to slow down the printing speed and send a warning to Octoprint. I've digged into the MCU_Commands list and I haven't seen anything to set up an interrupt driven action. Moreover, it is possible to schedule analog input reading but I haven't seen anything related to digital input reading at a given frequency, rather than the endstops implementation which is quite specific. Could please point me to the right direction?

KevinOConnor, 2019-05-25:
Take a look at klippy/extras/buttons.py and src/buttons.c.

-Kevin

smark-, 2019-05-29:
Thanks Kevin, I'm working on it!

closed 2019-05-29 by smark-