Klipper3d/klipper #1680: z minimum position

NicoSoftware, 2019-05-25:
Just a quick question, I dont find any setting in mesh-bed oder the probe section
I get this error:
```
Send: BED_MESH_CALIBRATE
Recv: // Failed to home probe: Timeout during endstop homing
Recv: // Make sure to home the printer before probing. If the probe
Recv: // did not move far enough to trigger, then consider reducing
Recv: // the Z axis minimum position so the probe can travel further
Recv: // (the Z minimum position can be negative).
Recv: !! Failed to home probe: Timeout during endstop homing
Recv: // Failed to home probe: Timeout during endstop homing
Recv: // Make sure to home the printer before probing. If the probe
Recv: // did not move far enough to trigger, then consider reducing
Recv: // the Z axis minimum position so the probe can travel further
Recv: // (the Z minimum position can be negative).
Recv: !! Failed to home probe: Timeout during endstop homing
```
Where can I find the z minimum position?
Thanks

theopensourcerer, 2019-05-25:
@NicoSoftware 

https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L41

NicoSoftware, 2019-05-25:
> @NicoSoftware
> 
> https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L41

thanks alot!
But thats the X stepper?

theopensourcerer, 2019-05-25:
@NicoSoftware It's an EXAMPLE config file. You can use these parameters in any of the main steppers, e.g. X Y and Z. Those which have end stops etc.... The value can also be negative.

theopensourcerer, 2019-05-25:
It is not a mandatory parameter. It has a default of 0.

NicoSoftware, 2019-05-25:
Yeah, just saw it, thanks alot, works perfect!

closed 2019-06-01 by klipper-gitissuebot