Klipper3d/klipper #1683: 'Command Format Mismatch' after last git pull.

angrymrtom, 2019-05-26:
did a `git pull` and then ran `install-octopi.sh`.  Running on ender3, now getting this error when connecting:

```
Command format mismatch: schedule_soft_pwm_out oid=%c clock=%u on_ticks=%u vs schedule_soft_pwm_out oid=%c clock=%u value=%hu
```

log attached
[klippy.log](https://github.com/KevinOConnor/klipper/files/3221311/klippy.log)

angrymrtom, 2019-05-27:
I removed my `klipper` folder, re forked, re flashed, and im not back up and running.  ill leave this open with the log if you wan't to look at anything.  Im not sure if there is anything you want to see or not.

KevinOConnor, 2019-05-27:
You should have seen advice in the OctoPrint terminal after receiving that error:
```
This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
```
See the FAQ for details on the full software upgrade procedure.

-Kevin

angrymrtom, 2019-05-27:
sounds good, thanks.

closed 2019-05-27 by angrymrtom