Klipper3d/klipper #1685: PID tune way off.

thucar, 2019-05-28:
I've been struggling with my hotend PID tune for a while now. I can not seem to be able to get a stable result. The temperature fluctuates a lot to the point where I need to wait for about 10 minutes at the beginning of each print to get it to register that the target has been reached.

During print, the temperature fluctuates from 247C to 265C when my target is 255C
I'm using a 40W heater and these are my results from two separate PID calibrations.

```
pid_kp = 23.876
pid_ki = 1.162
pid_kd = 122.660

pid_Kp=19.510
pid_Ki=0.963
pid_Kd=98.769
```

[klippy.log](https://github.com/KevinOConnor/klipper/files/3225692/klippy.log)

terrorhai, 2019-05-29:
Which board do you use? Is the correct thermistor type applied in your config?

thucar, 2019-05-29:
I'm using a RAMPS 1.4 and the thermistor type should be set in the config correctly: `sensor_type = ATC Semitec 104GT-2`

terrorhai, 2019-05-29:
Okay, 1. thermistor is dead or 2. on the MKS Gen L (which is literally a ramps + mega 2560 in one board) at 24v there is a problem with hardcore ripple which makes PID controlling very unstable. But I don't know how a standard ramps is reacting to 24v. I would change the thermistor first.

thucar, 2019-05-30:
I doubt it is a hardware issue. I was running my system on 12V up until about a month ago and was having the same issue. Now when upgrading to 24V I swapped out the hotend in its entirety - the heater block as well as the thermistor.

Looking at the log file it is clearly visible that the heater is being operated basically "on/off" and there's not much of pwm cycle changing going on. The duty cycle is either 1.000 or 0.000

```
Stats 3482.9: gcodein=2431 mcu: mcu_awake=0.014 mcu_task_avg=0.000114 mcu_task_stddev=0.000083 bytes_write=417694 bytes_read=361630 bytes_retransmit=0 bytes_invalid=0 send_seq=19017 receive_seq=19017 retransmit_seq=0 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=9 stalled_bytes=0 freq=15902141 heater_bed: target=110 temp=109.9 pwm=0.853 print_time=2358.916 buffer_time=0.000 print_stall=0 extruder: target=255 temp=257.1 pwm=1.000
Stats 3483.9: gcodein=2431 mcu: mcu_awake=0.014 mcu_task_avg=0.000114 mcu_task_stddev=0.000083 bytes_write=417832 bytes_read=361775 bytes_retransmit=0 bytes_invalid=0 send_seq=19027 receive_seq=19027 retransmit_seq=0 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15902141 heater_bed: target=110 temp=109.6 pwm=1.000 print_time=2359.910 buffer_time=0.000 print_stall=0 extruder: target=255 temp=266.6 pwm=0.000
Stats 3484.9: gcodein=2431 mcu: mcu_awake=0.014 mcu_task_avg=0.000114 mcu_task_stddev=0.000083 bytes_write=417944 bytes_read=361910 bytes_retransmit=0 bytes_invalid=0 send_seq=19035 receive_seq=19035 retransmit_seq=0 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15902141 heater_bed: target=110 temp=111.0 pwm=0.000 print_time=2360.904 buffer_time=0.000 print_stall=0 extruder: target=255 temp=253.8 pwm=1.000
Stats 3485.9: gcodein=2431 mcu: mcu_awake=0.014 mcu_task_avg=0.000114 mcu_task_stddev=0.000083 bytes_write=418073 bytes_read=362069 bytes_retransmit=0 bytes_invalid=0 send_seq=19045 receive_seq=19045 retransmit_seq=0 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15902140 heater_bed: target=110 temp=110.7 pwm=0.000 print_time=2361.899 buffer_time=0.000 print_stall=0 extruder: target=255 temp=264.9 pwm=0.000
Stats 3486.9: gcodein=2431 mcu: mcu_awake=0.015 mcu_task_avg=0.000122 mcu_task_stddev=0.000085 bytes_write=418219 bytes_read=362248 bytes_retransmit=0 bytes_invalid=0 send_seq=19056 receive_seq=19056 retransmit_seq=0 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15902140 heater_bed: target=110 temp=109.5 pwm=1.000 print_time=2362.894 buffer_time=0.000 print_stall=0 extruder: target=255 temp=264.6 pwm=0.000
Stats 3487.9: gcodein=2431 mcu: mcu_awake=0.015 mcu_task_avg=0.000122 mcu_task_stddev=0.000085 bytes_write=418342 bytes_read=362393 bytes_retransmit=0 bytes_invalid=0 send_seq=19066 receive_seq=19066 retransmit_seq=0 srtt=0.003 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15902139 heater_bed: target=110 temp=109.8 pwm=0.709 print_time=2363.888 buffer_time=0.000 print_stall=0 extruder: target=255 temp=252.9 pwm=1.000
```

KevinOConnor, 2019-05-30:
The thermistor is not providing reliable readings - the temperature is jumping +-10 degrees from second to second.  I'd double check the thermistor wiring, and/or replace the thermistor.

-Kevin

BlackStump, 2019-05-30:
or check for a dry joint on the ramps if the ramps is the common denominator

DanielJoyce, 2019-06-07:
Or thermistor is loose in hotend or poorly seated or weak wire is coming apart.

thucar, 2019-06-07:
I've swapped out the thermistor for another one but the issue remains. However, I did notice something. The problem only manifests if I have the bed heater on at the same time AND the bed has reached and is maintaining its target temperature.

So:
- bed heater off: no issues
- bed heater at 100% duty cycle: no issues
- bed heater duty cycle between 0% and 100%: hotend thermistor readings fluctuate.

BlackStump, 2019-06-07:
check your power supply, maybe under powered or low quality noise ripple, check for voltage fluctuations
voltage at no bed heater
voltage at 100% duty cycle to bed
Then I suspect voltage changing as bed cycles

KevinOConnor, 2019-06-18:
Your latest symptoms indicate a voltage supply issue.  Any further updates?

-Kevin

thucar, 2019-06-18:
I am currently waiting for some more thermistors to test with. The two I have, both behave in the same manner. 

I did measure the PSU voltage with all heaters off, with only bed heater at 100% and with both, the bed and hotend heater at 100%. The output is rock solid at 24.5V at all times.

mental405, 2019-06-19:
The voltage ripple may be occurring on the RAMPS board itself. I had very similar issues on two of my boards and upon hooking a scope up to the 5V pin you could see a very defined oscillation in the voltage. I had to add additional decoupling capacitors to the board to get it stabilized.

KevinOConnor, 2019-07-10:
Any further updates?

-Kevin

KevinOConnor, 2019-07-29:
I'm going to close this issue as it appears the conversation has concluded.

-Kevin

closed 2019-07-29 by KevinOConnor