Klipper3d/klipper #1686: Option 'driver_SGT' in section 'tmc5160 stepper_x' must have minimum of 0

Oranginator, 2019-05-28:
Hi there,

I was tuning my stallguard and wanted to go down beyond 0 like -5, which should be possible as https://github.com/KevinOConnor/klipper/blob/master/docs/Sensorless_Homing.md says a range between -64 and 63 is possible.
My printer didn't trigger stallguard with a value of 0, I'm using 0.9° steppers.

After reloading the firmware I'm getting the error: 

> Option 'driver_SGT' in section 'tmc5160 stepper_x' must have minimum of 0
> Once the underlying issue is corrected, use the "RESTART"
> command to reload the config and restart the host software.
> Printer is halted

[klippy.log](https://github.com/KevinOConnor/klipper/files/3225926/klippy.log)

DBa2016, 2019-05-28:
Did you check your wiring?

Oranginator, 2019-05-28:
Wires are okay.

FHeilmann, 2019-05-28:
You are setting the content of the register. The register value (for the sake of simplicity) is always a positive value and you have to add 64 to your desired 'real' SGT value to obtain the register contents.  A value of 0 in driver_SGT actually corresponds to -64. So in order to set a 'real' SGT value of -5, you have to configure driver_SGT as 61 in your klipper config. This can go up to 127, which would correspond to a 'real' SGT value of 63.

Oranginator, 2019-05-28:
Thank you I'll try again later and check some things.

KevinOConnor, 2019-05-28:
@FHeilmann - FYI, the SGT field is actually in two's complement format.  So, 0 is 0, 1 is 1, 2 is 2, etc.  However, -1 is 127, -2 is 126, etc.  We should probably update FieldHelper to handle this as it's not really intuitive.  (FWIW, tmc2130.py does handle this for driver_SGT, but doesn't handle it in SET_TMC_FIELD.)

-Kevin

FHeilmann, 2019-05-28:
@KevinOConnor I see. In that case we should check the 2660 implementation and the 5160 implementation. I know that the 2660 is faulty, and maybe @Stephan3 wants two recheck his implementation as well.

Stephan3, 2019-05-28:
i will lokk into that, thanks for the pointer.

KevinOConnor, 2019-05-29:
I think this should be fixed now (commit 6903ab87).

-Kevin

Oranginator, 2019-05-29:
Thank you I'll try later and give you some feedback.

closed 2019-05-30 by Oranginator