Klipper3d/klipper #1687: Stop-Behavior Could Be Improved

VanKurt, 2019-05-28:
When hitting the stop button in OctoPrint the printer stops as expected. After that I usually want to move the nozzle away from the print as fast as possible, so that I can clean up the mess.

**The problem is:**
After hitting stop, Klipper looses its homing. Thus I cannot issue any move commands. Instead I have to turn the printer off and start rotating the z lead screw by hand.

**It would be better if:**

- The printer would still react to move commands
- The printer would move away from the print automatically (e.g. lift by 20mm)
   (**or even better**: execute a custom GCODE from the config file when stopping)

Is there any way I can achieve that behavior via the configuration?
If not: would it be possible to fix that in an upcoming Klipper release?

PS: I cannot attach a log file since Klipper does not seem to create one (I checked /tmp and /var/log). Also issuing a M112 does not help.

theopensourcerer, 2019-05-28:
Create a gcode macro.

FHeilmann, 2019-05-28:
Your stop script in octoprint likely has a M84 in it which disables the motors. Remove that and you should be able to move the head out of the way. Better yet, add the gcode to move the head out of the way to your octoprint stop script to execute it automatically.

Hywelmartin, 2019-05-29:
not a klipper issue
I have added a G28, but if you don't home to max you could home x, y or both before M84
[image]

VanKurt, 2019-05-29:
OK, I'll try adjust the stop script in OctoPrint like this:
```
G91 ; Relative positioning
G1 Z20 ; Move nozzle up 20mm
G28 X Y ; Home X and Y
M84 ; Disable steppers
```

Thanks!

closed 2019-05-29 by VanKurt