Klipper3d/klipper #1688: Bed Levelling Mesh Ignored While Printing

VanKurt, 2019-05-28:
I recently tried to use automatic bed levlling with a BLTouch. To so I performed the following steps:

- Added BLTouch and levelling mesh sections to the Klipper config
- Used the "Bed Visualizer" OctoPrint plugin to make sure the probing works as expected
- Stored the reported mesh to the Klipper config
- Updated my Cura start GCODE to load the previously saved mesh

The probing itself looks fine. The reported heights seem reasonable and the mesh is stored correctly in my Klipper config.

**The problem:**
During the actual print the levelling info seems to be completely ignored. I.e. the nozzle is sometimes so close to the bed that it almost starts scratching the glass. In other areas it is so far away that the filament won't stick to the bed.

**Attached files:**
Printer config: [printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/3226196/printer.cfg.txt)
Terminal log from Octoprint: [PrintLogTerminal.txt](https://github.com/KevinOConnor/klipper/files/3226185/PrintLogTerminal.txt)
Cura start GCODE: [CuraStartGCODE.txt](https://github.com/KevinOConnor/klipper/files/3226187/CuraStartGCODE.txt)

Picture of the test print:
[image]

NicoSoftware, 2019-05-30:
Did you get this fixed?

VanKurt, 2019-05-30:
No, sadly not. Currently I have no clue what may cause this.

KevinOConnor, 2019-05-30:
You'll need to attach the Klipper log file.  Without the log there is little we can do.

-Kevin

VanKurt, 2019-05-30:
There is no log file :-(
I checked /tmp and /var/log via ssh and CyberDuck. No .log file in sight. I also tried to do a M112, restart Klipper host and the whole RaspberryPi without success. Is there anything I can do to force the creation of a log?

KevinOConnor, 2019-05-30:
How was the Klipper software installed?  If you ssh into the rpi, can you cut-and-paste the full output from `ls -l /tmp/`?

-Kevin

VanKurt, 2019-05-31:
OK, I restarted everything a couple of times more and started a new print.
Now there even two log files :-)

[klippy.log](https://github.com/KevinOConnor/klipper/files/3239845/klippy.log)
[klippy.log.2019-05-30.log](https://github.com/KevinOConnor/klipper/files/3239848/klippy.log.2019-05-30.log)

Arksine, 2019-05-31:
So just looking at your log there is very little adjustment to be made.  Assuming the origin is at the front left, that corner is .1mm higher, the right rear is up to .12mm higher.  The rest of the bed is within .05mm.  Are you calibrating at print temperature?

VanKurt, 2019-06-01:
> So just looking at your log there is very little adjustment to be made
The z offset calibration is absolutely fine. The only problem is that the the z offset varies quite a lot across the bed. In some areas the nozle gets too close to the bed, in others its too far away. I thought this would be fixed by using the bed-height-mesh-levelling-feature-thingy, but as my photo above shows it does not. The mesh seems to be not applied at all.

xelprep, 2019-06-01:
Perhaps you could try increasing the mesh grid to 7,7 and/or changing the algorithm to bicubic? Not sure if this will help you but it's what I use on mine and I can see on the LCD that the Z axis appears to be changing to follow the mesh as X and Y are moving on the same layer.

Arksine, 2019-06-01:
@VanKurt I was referring to doing BED_MESH_CALIBRATE at the bed's print temperature, not the z-offset.  As i said, the probed points show very little adjustment.  The bed_mesh module isn't getting good input from the probing procedure.

VanKurt, 2019-06-03:
I changed the algorithm to bicubic and repeated my tests. Now everything works as expected :-)

closed 2019-06-03 by VanKurt