Klipper3d/klipper #1690: SKR v1.3 MAX6675 and 60W heater

Geramy, 2019-05-30:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3235424/klippy.log)

I just purchased brand new thermocouple and heater unit for this 3D printer and after trying to get it to print it keeps heating up and starting to print then after about 5 minutes it will stop and say minimum temperature below requirement, and quit printing. which that's not a bad thing the bad thing is that if I try to print the temperature lowers for some reason it won't stay up like it's supposed to. If I set the printers heater to 215 it will stay there for more then 30 minutes but if I print it literally stops heating.

Geramy, 2019-05-30:
Sorry this ticket is invalid I looked at the G-Code from cua engine and it seems that with the plugin for multiple extruders it actually sets the T1 tool temp to 0 during the print. Is there any better slicing tool available to use with octoprint?

BlackStump, 2019-05-30:
Octoprint is a gcode sender and you can use any slicer you want and upload it to Octoprint
Cura
Slic3r
ideamaker
craftware
simplified3d
icesl
kisslicer
there is most likely some I have missed but plenty to choose from

KevinOConnor, 2019-06-09:
I'm going to close this issue as it looks like the original issue reported is no longer thought to be an issue.

-Kevin

closed 2019-06-09 by KevinOConnor