Klipper3d/klipper #1691: advanced delta calibrate worse

vader333, 2019-05-30:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3236151/klippy.log)
I've been trying to get my delta dimensional accuracy to improve, but it seems the advanced delta calibration routine makes things worse. I've tried two procedures

1. delta_calibrate, bed_mesh_calibrate, print calibration piece, delta_analyze calibrate=extended, delta_calibrate, bed_mesh_calibrate
2. delta_calibrate, bed_mesh_calibrate, print calibration piece, delta_analyze calibrate=extended, bed_mesh_calibrate

Both give me progressively worse results.

Table of results for center distances:
 A      | B        | A-74 | B-74
____________________________
73.93 | 73.63 | -0.07 | -0.37
74.15 | 73.83 | 0.15 | -0.17
73.95 | 73.72 | -0.05 | -0.28
73.89 | 73.79 | -0.11 | -0.21
73.95 | 73.72 | -0.05 | -0.28
73.8 | 73.82 | -0.2 | -0.18

RMS error worsens by 0.341055

0.290689 | 0.631744

My rods are tied together with timing belts so backlash is minimal, and my probe repeatability is 0.05. I have not measured my delta rods to ensure they are equal length, but they seem pretty alright. There is minimal tower angle offset, I've checked that with a separate delta calibration print without the advanced delta calibrate parameters.

Is there anything in the hardware that might make the calibration unusable that i should check?

BlackStump, 2019-05-30:
what sort of probe do you have?
If it is offset in anyway from the nozzle it could be suffering from effector tilt but if you followed Kevin's instructions he makes mention of that and what he recommends to avoid those introduced errors.
All firmware's that have delta calibrate have some assumptions and some of them are: 
The bed is perfectly flat.
The bed is a at perfect right angle to all three towers
The arms are at perfect lengths to each other and it does make it easier if they been measured precisely hinge to hinge.
I could go on but you get the picture. if you have problems then a manual old school calibrate is the best starting place.
There is plenty of tutorials on the net to follow.
Or follow Kevin's instructions and do the manual paper test calibrate.
For probes on a Delta I would highly recommend a Piezo that uses the nozzle as the probe they do not suffer from effector tilt.
I also don,t think it is a good idea to use bed mesh while doing delta calibrate.

vader333, 2019-05-30:
I'm using anycubic's endstop probe on my Anycubic predator. It's a fairly good probe with high repeatability, and it sits right underneath the nozzle, so effector tilt isn't a concern.

I've tried manual calibration on my other delta's in the past, and it was an exercise in futility. Each probing took so long, and the result was just as bad, if not worse.

I'm familiar with the delta calibration assumptions, but I'm not sure if there's a good way to measure them without specialised tools. The bed is pretty flat, and the towers are at 120 degrees from each other.  The only other thing I can do in the near future is to measure my rods, but that's going to be a bit of a hassle, I was hoping there might be a software solution.

KevinOConnor, 2019-05-30:
FWIW, the probe that came with my delta printer is very repeatable.  Unfortunately, it's repeatedly incorrect.  When the effector has a tilt, the distance between probe/bed and the distance between nozzle/bed changes.  So, the probe is accurate, it just is no longer measuring anything useful.

To determine if the probe has this type of systemic error, upgrade to the latest code version (see the FAQ for upgrade details) and run the PROBE_CALIBRATE command as described at https://www.klipper3d.org/Probe_Calibrate.html .  It's important to go through the full "paper test" steps using the TESTZ command as described in that document.  After running the test note the resulting z_offset.  Then move the head to a position near the A tower and rerun the full PROBE_CALIBRATE test again.  Then move to a position near the B tower, run PROBE_CALIBRATE, then near the C tower and run PROBE_CALIBRATE.  The maximum difference between any of the four calculated z_offset values must be within about 15microns (.015mm).  If it is not, then alas, the probe isn't going to be accurate enough - use manual delta calibration instead.

It's difficult to mix delta calibration with bed mesh.  I'm not surprised you've gotten poor results with it.  The delta calibration requires a flat bed to work correctly.  If delta calibrate doesn't provide good results, then bed mesh isn't going to improve things.  It is possible to use bed mesh on a delta, but it only makes sense to do that after the delta parameters have been calibrated to excellent precision.  Otherwise, the bed mesh will just amplify the error from inaccurate delta parameters.

-Kevin

vader333, 2019-05-31:
I see! That's very useful information. I also realise I've been measuring the error wrongly. The RMS error and maximum error both have been decreasing with each successive calibration. I guess I was just tardy in my recording, but I'm glad I asked.

Thanks for your answer, I'll definitely check that and see what kind of performance I get.

closed 2019-05-31 by vader333

sprig, 2020-09-27:
@vader333 I realize this is an old ticket. are you / have you been using stock hardware Predator? If so, did you end up getting results good results? Would you care to share your config?

Thanks!