Klipper3d/klipper #1693: Intermittent Timer too close

PreownedFIN, 2019-06-01:
Here's the klippy log file:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/3243629/klippy.zip)
Started a print and the printer got almost to the printing temperature, before erroring out.

I'd been running Klipper for my Anycubic Kossel Linear+ inside my Octoprint VM for months without a problem. Now after a weeklong cold period (which may include the VM Ubuntu updates, not 100% sure), I started to encounter these "Timer too close"-errors.
Sometimes the error pops up as soon as I connect to the printer, sometimes while heating up for a print, sometimes when the printer is just idling, and sometimes during a print. (I did have a quick ~30min print that managed to finish just fine in the middle of all this.)

I've had this kind of errors before, but previously they were fixed with something like changing the cable or by restarting the VM or the printer.

I played around with changing the cables, replacing the Trigorilla board, rebooting the VM and the VM host, changing the VM- and process affinity. Nothing seems to have worked.

I didn't really capture logs, as I was working on the assumption *something* is awry on the hardware side.
Before I start breaking out the Raspis and such, I thought that I'd post my current progress here. Hopefully I've just overlooked something obvious that shows in the klippy logs.

As for the logs, there doesn't seem to be *that* many retransmits. There are some, but not that many that would (at least I think) cause too long of an interruption.

So, anything that I might have missed? 😅

KevinOConnor, 2019-06-02:
The log indicates the host software "pid thread" didn't get any cpu time for over 500ms.  That resulted in a failure.

It looks like you've implemented a custom install of the host software.  It's necessary to ensure the host environment is stable enough so that the software has some basic responsiveness (no other high priority processes that starve the cpu, nothing pushing the machine into swap, etc.).

-Kevin

PreownedFIN, 2019-06-02:
Thanks for the info!

My setup is quite "custom", and has its quirks. Starving threads on VMs hasn't been one of them.
I played around with the CPU provisioning and the niceness of the Klipper python thread, neither of which helped.

I restored a snapshot from before the updates and the problem hasn't manifested itself, **yet**.
I'll keep the "broken" snapshot running for my 'beater-printer' and I'll see if I can find a reason for the resource starvation.

It's been a couple of years since my last "python-thread-starvation"-hunt, but something to ponder over for a while.

And just to make sure that I interpreted the klippy log properly:
The timings jump (assumingly)~700ms between the "Sent 97" and "Sent 98" lines.
And that is indicative of "nothing" happening for that interval? (The lack of CPU time.)
The 2 'timings' *seem* to indicate some kind of duration. The time it took to process the message?
```
Sent 94 174.323416 174.322816 15: seq: 13, schedule_soft_pwm_out oid=19 clock=1585408000 on_ticks=382060
Sent 95 174.382137 174.381897 6: seq: 14, get_clock
Sent 96 174.503226 174.502706 13: seq: 15, schedule_digital_out oid=7 clock=1588287999 value=0
Sent 97 174.623334 174.622734 15: seq: 16, schedule_soft_pwm_out oid=19 clock=1590207999 on_ticks=481270  
Sent 98 175.367189 175.366949 6: seq: 17, get_clock  
Sent 99 176.199274 176.198274 25: seq: 18, schedule_soft_pwm_out oid=19 clock=1599808000 on_ticks=649776, schedule_soft_pwm_out oid=19 clock=1604608000 on_ticks=550170
```

As far as I'm concerned, this issue can be closed.
If I find a culprit or even a resolution, I'll add them under this issue.

And again, thank you for showing the general direction of the problem! :)

KevinOConnor, 2019-06-02:
If you run the log file through `scripts/logextract.py` (see docs/Debugging.md for the details), you'll find:

```
Receive: 94 1254450.115606 1254449.900379 14: seq: 12(289), analog_in_state oid=6 next_clock=1832960000(1254450.406778) value=6765
Sent 98 1254450.216722 1254450.216482 6: seq: 12(290), get_clock
Receive: 95 1254450.217750 1254450.216722 11: seq: 13(290), clock clock=1829913944(1254450.216291) 
Receive: 96 1254450.236633 1254450.216722 14: seq: 13(290), analog_in_state oid=18 next_clock=1834880000(1254450.526847) value=7225
Receive: 97 1254450.416670 1254450.216722 14: seq: 13(290), analog_in_state oid=6 next_clock=1837760000(1254450.706951) value=6767
Receive: 98 1254450.536737 1254450.216722 14: seq: 13(290), analog_in_state oid=18 next_clock=1839680000(1254450.827019) value=7190
Sent 99 1254450.635619 1254450.635099 13: seq: 13(291), schedule_digital_out oid=7 clock=1833088000(1254450.414783) value=0
Receive: 99 1254450.637805 1254450.635619 12: seq: 14(291), shutdown clock=1836613004(1254450.635222) static_string_id=Timer too close
MCU 'mcu' shutdown: Timer too close
```

The "Receive 94" line is what led to the "Sent 99" line.  It appears from the timing that the serialqueue.c C thread is running nominally (the timestamps look okay), but when the messages are handed off to the "python pid thread" the reaction times are absurdly high.  Normally the pid thread would respond within a few milliseconds.

The above is my conclusion from the analysis - it's always possible there's a different explanation.

-Kevin

KevinOConnor, 2019-06-18:
Okay, I'm going to close this issue - I can reopen if you have further information or questions.

-Kevin

closed 2019-06-18 by KevinOConnor