Klipper3d/klipper #1695: [bug, feature.. i'm not sure...] Pause inside a macro activates at end of macro

Hywelmartin, 2019-06-01:
I have a start script that calls a start macro. 

I'm not sure if this is a bug or not, maybe something to mention in the document for pause/resume...

This macro is rather long and have the purge line in that. [video](https://youtu.be/hXh3WveA_4o) and i the beginning I have a PAUSE where I load filament..

What doesn't work
```
[gcode_macro THE_START]
gcode:
    SET_PIN PIN=led VALUE=1
    G28
    G21
    M140 S{B}
    M104 S150
    G90
    M117 HERE WE GO..
    G28
    M190 S{B}
    M109 S160
    NOZZLE_CLEAN
    set_delta_height
    G1 X-74 Y-43 Z50 F3000
    PARK_MACRO X=-74 Y=-43 Z=10
    M104 S{H}
    M109 S{H}
    PAUSE_MACRO
    G92 E0
    G1 X-44 Y-75 Z0.6 E3 F3000
    M117 SQUIRTING...
    G1 X2.183 Y-95.623 E10 F500
    LA_LINEA
    G1 E-0.4 F500
    M204 S3000
    G1 X87 Y-17 Z1 F8000
    G1 Z5
    M117 DOODELING...
```
If I set up my start gcode in the slicer like:
THE_START B=[first_layer_bed_temperature] H=[first_layer_temperature]
THE_START_END
IT WORKS!!!!
```
[gcode_macro THE_START]
gcode:
	SET_PIN PIN=led VALUE=1
	G28
	G21
	M140 S{B}
	M104 S160
	G90
	M117 HERE WE GO..
	G28
	M190 S{B}
	M109 S160
	NOZZLE_CLEAN
	set_delta_height
	G1 X-74 Y-43 Z50 F3000
	PARK_MACRO X=-74 Y=-43 Z=10
	M104 S{H}
	M109 S{H}
	PAUSE_MACRO
```
	
```
[gcode_macro THE_START_END]
gcode:
	G1 X-74 Y-43 Z50 F3000
	G92 E0
	G1 X-44 Y-75 Z0.6 E3 F3000
	M117 SQUIRTING...
	G1 X2.183 Y-95.623 E10 F500
	LA_LINEA
	G1 E-0.4 F500
	M204 S3000
	G1 X87 Y-17 Z1 F8000
	G1 Z5
	M117 DOODELING...
```
LOG
[klippy(STARTSCRIPT).txt](https://github.com/KevinOConnor/klipper/files/3244070/klippy.STARTSCRIPT.txt)

Arksine, 2019-06-02:
This is expected behavior.  The PAUSE command is intended to stop octoprint or the SD card from sending further gcode.   It will not pause execution of gcodes after a PAUSE inside of a macro.

Hywelmartin, 2019-06-02:
Ok
Thanks for clarification..

by the way, if you didn't know, if the pause/resume send a //action:paused or //action:resumed octoprint behaves... much nicer... [link](http://docs.octoprint.org/en/master/features/action_commands.html)

Arksine, 2019-06-02:
I see.  These must be new action commands, as I don't recall seeing them before.  I'll update the module to support this.

closed 2019-06-02 by Hywelmartin