Klipper3d/klipper #1701: Klipper "home" timeout not logged

adwsail, 2019-06-04:
I have klipper running nicely on a CR10S. No problem with operation. However when calibrating the step size for x and y axis I noticed that after homing then issuing a command, say G1 X50 Y50 it will move there as expected. Five minutes later, with no input at all, just making notes, klipper will refuse to move saying It must be homed first. This is repeatable every time. Home the machine, execute a move, wait 5 minutes and you MUST rehome. Is there any way to modify this firware timeout? No issue printing or in general operation as long as its allowed to sit for a few minutes you will be forced to rehome.

Its not a deal breaker or even a problem, just a minor inconvenience. Nothing unexpected in klippy,serial or octoprint logs because there has been no error or fault. It just times out. No heating or any other operation is in process, just sitting while I write my notes. Any ideas appreciated.

jakep82, 2019-06-04:
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L477

adwsail, 2019-06-04:
Perfect answer, greatly appreciated. Case closed.

KevinOConnor, 2019-06-09:
I'm going to close this issue as it looks like a solution was found.

-Kevin

closed 2019-06-09 by KevinOConnor