Klipper3d/klipper #1702: TMC driver UART init order, can't send gcode_macro

vstruk, 2019-06-04:
I have the following setup:

- RaspberriPi powered by 5v source
- Arduino is powered by USB through RaspberryPi.
- RAMPS(with a diode D1 removed) and all the rest are powered by a discrete 24v PSU.

This discrete PSU supposed to be controlled by arduino PIN: 
"gcode: SET_PIN PIN=main_psu VALUE=1"

It works as expected with a 'standalone' stepper driver configuration.
The command fails once I enable 'single wire uart mode'(see the config below).
SET_PIN just returns the error with no effect on the pin state:

`
Send: SET_PIN PIN=main_psu VALUE=1

Recv: // Unable to read tmc2208 'stepper_x' register IFCNT
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Printer is haltedRecv: !! Unable to read tmc2208 'stepper_x' register IFCNT
`

Seems like driver initialization via UART occurs too early.
That makes impossible to use UART-enabled drivers with an external PSU setup.

I'm using a TMC2208 with single wire uart:

[tmc2208 stepper_x]
uart_pin: ar63
microsteps: 16
run_current: 0.6
stealthchop_threshold: 0

[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
....

jakep82, 2019-06-04:
I'm not sure this is a safe way to power up the board.  I believe TMC drivers need to have motor voltage (24V) before they receive logic voltage.

vstruk, 2019-06-04:
> I'm not sure this is a safe way to power up the board. I believe TMC drivers need to have motor voltage (24V) before they receive logic voltage.
Is that statement based on a datasheet or any other document?
Could you please share a link?

vstruk, 2019-06-04:
+klippy.log
[klippy.log](https://github.com/KevinOConnor/klipper/files/3254779/klippy.log)

mjedlinski, 2019-06-05:
https://learn.watterott.com/silentstepstick/faq/

amblidex, 2019-06-07:
The problem is that the TMC drivers require motor power first, then a delay of a few hundred ms before initialization or else you see the error like you are experiencing. The workaround is to live with Klipper halting the first time you start it, then issuing a firmware restart command in the terminal to restart everything again (and this time the motor voltage will already be up and running).

A cleaner setup is possible with the TMC2130 because it will not halt Klipper on startup if the init fails. I use gcode scripts with INIT_TMC and a delay and have Octoprint manage the power. The TMC2108 driver however, does not allow startup with a failed init because of: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#my-tmc-motor-driver-turns-off-in-the-middle-of-a-print

vstruk, 2019-06-07:
amblidex, thanks for you answer.

I think you didn't quite understand what I mean.
Let me try to explain it in other words.                                                                                                                                                                            

What I want to achieve is to keep the main PSU (along with heaters&other stuff) off while printer is in standby.
It should be off at the moment Rpi and Klipper starts as well.
'shutdown_value' and 'value' will be set to '0' for the pin controlling PSU. So the power supposed to be shut off instantly in case of any error.
Just as another safety precaution.

That's ok to wait some time after a power-on and then issue a firmware restart command.
The thing is I just can't control the pin and therefore can't turn on PSU at all while drivers are configured to use UART. Klipper just ignores 'SET_PIN' command and responds with an error message.
On the other hand it works perfectly while 2208's are in standalone mode.

That seems like a bug.

amblidex, 2019-06-07:
Yep, I tried to set up mine the same way, so that the main power is only on when I'm printing. But Klipper's 2208 driver does not allow klipper to start if it can't init the chip because of the the problem I linked to above. You'd have to take it up with @KevinOConnor to change that behavior. Maybe he would be willing to allow an option in the driver that does not require init to succeed at startup.

jakep82, 2019-06-07:
You could use a gpio pin from the Pi to control the PSU. I believe Octoprint has a plug-in that allows control of the RPi going through gcode.

vstruk, 2019-06-08:
Right, but that's a workaround and not a good one.
As the entire setup with 2nd PSU built because of safety concerns, I'd like to rely on a solution that is close to hardware as much as possible. 
Klipper firmware on the controller side look much better than a stack of Rpi+linux+python+octoprint

jakep82, 2019-06-08:
You need all of those things plus firmware on the MCU to use gpio on the controller. I'm not sure how it's any different.

vstruk, 2019-06-08:
Nope. Once firmware goes down (any reason: failed sensor, thermal runaway, whatever) -> power shuts down as well, almost instantly(pin has shutdown_value: 0)
"All those things" are just to control PSU under normal conditions. That's the difference.

vstruk, 2019-06-08:
Here is the solution I've come up:

1) Disable auto init(see the PR above)
2) Configure the following macro:

```
[output_pin main_psu]
pin: ar57
shutdown_value: 0
value: 0

# power on                                                                                                                                                                                                          
[gcode_macro M80]                                                                                                                                                                                                   
gcode:                                                                                                                                                                                                              
  SET_PIN PIN=main_psu VALUE=1                                                                                                                                                                                     
  G4 P2500
  INIT_TMC stepper=stepper_x
  INIT_TMC stepper=stepper_y
  INIT_TMC stepper=stepper_z

# power off
[gcode_macro M81]
gcode: SET_PIN PIN=main_psu VALUE=0
```

KevinOConnor, 2019-06-12:
If, on an init failure, Klipper waited a few seconds and retried - would that address the issue you are facing?

-Kevin

amblidex, 2019-06-12:
That should solve the problem, and the retry delay need only be a few hundred milliseconds.

However, this in itself will only fix the 2208 problem because the 2130 (and probably the other SPI TMCs) fail silently on init. For SPI TMC, maybe we can also try to detect a failed init by reading and checking the register state before continuing.

KevinOConnor, 2019-06-12:
I agree that the tmc2130 code should check register writes.  We can do that by verifying the spi response from the write message.  (Unfortunately, it will break the config for people that haven't wired the MISO pin correctly, but it's probably worth that pain.)

-Kevin

vstruk, 2019-06-17:
> If, on an init failure, Klipper waited a few seconds and retried - would that address the issue you are facing?

No, that won't help. Driver will stay powered off(and won't respond, obviously) until I turn on the main PSU. That can happen hours after Rpi and klipper had started.
I resolved the issue with the patch (see PR above) and it works perfectly for me.
You can either accept the PR or wait until you finish PR #1707 and PR #1710(as you suggested in a comment to my PR).

Here is what you wrote:

> Also, after those two PRs are merged, it may be possible to delay tmc init until the first "printing command" is received. That may avoid the need for a separate config option.

KevinOConnor, 2019-06-18:
PR #1707 was merged and it is the only dependency.

Unfortunately, after further review, it does not seem practical to send the init sequence during the first "printing command".  We can change the code to do that, but if the init fails then there isn't a good way to report the error.  Given the severity of that error I think the error handling needs to be solid.

In contrast, retrying on a failed init is trivial to implement and is harmless.

I understand that this solution may not work for you.  However, given that the trinamic drivers clearly state that motor power must be provided prior to powering their digital interfaces, my feedback would be that it probably does not make sense to add code to the main Klipper repo to support it.

-Kevin

vstruk, 2019-06-27:
> trinamic drivers clearly state that motor power must be provided prior to powering their digital interface

Documentation states it is  "only applicable for SilentStepSticks with variable 3-5V logic voltage".
I'm not sure about the drivers i'm using but i've been running this patch for about ~250 hours and a lot of power cycles. Drivers are fine.

I'd propose to add this patch and provide example config with a warning note like "Be careful, this is not recommended by the driver vendor".  I could do this.

Will you merge it?

KevinOConnor, 2019-06-28:
For the main Klipper repo, I think the code should somehow validate that the init sequence has been sent prior to movement.  I fear the consequences of not sending an init would be too severe (using random current, incorrect step size, tmc2208 stealthchop power off problem).  It would be risky for the user and I fear a maintenance burden on support.

-Kevin

vstruk, 2019-06-28:
Well, it make sense. 
But on the other hand it is better to have an option to configure behavior you need(and understand possible consequences) than not having it at all.
I believe flexibility worth it.

Warning message can be improved like "This is potentially harmful option. Use it only if fully understand what are you doing. Driver MUST be initialized manually if the option is enabled. Possible consequences of not sending init sequence are: using random current, incorrect step size, tmc2208 stealthchop power off problems or even dead TMC driver"

KevinOConnor, 2019-06-30:
Thanks for the feedback.  Unfortunately, I don't think documentation or warnings is sufficient.  Too many users don't read the documentation - for those few that do, I don't think it make sense to burden them with config documentation that effectively says "don't use this config option".

FWIW, I regularly disable motor power to my delta printer which has tmc2208 drivers.  I also regularly get the tmc2208 init error.  It's never been a problem for me, though, as I just enable power and then issue a RESTART command.  I'm not sure why that solution doesn't work for you.

That said, I'm certainly open to an automated solution - it would even make things simpler for me.  I do think the main Klipper code should continue to have basic sanity checks prior to movement, though.  (And, individuals that know what they are doing, can modify the host Python code accordingly.)

-Kevin

KevinOConnor, 2019-07-01:
Thinking about this further, one possible solution would be to send the init sequence on a "idle_timeout:printing" event and if that fails for some reason, then instruct the firmware to enter a shutdown state.  This would not be ideal error handling, but it should at least be safe.

I'm curious, though, why enabling the power supply during startup doesn't work for you (`[output_pin power_supply_enable] value: 1`)?  I would think that would enable power at startup, which would allow the initial configuration to proceed normally.

-Kevin

vstruk, 2019-07-04:
> FWIW, I regularly disable motor power to my delta printer which has tmc2208 drivers. I also regularly get the tmc2208 init error. It's never been a problem for me, though, as I just enable power and then issue a RESTART command. I'm not sure why that solution doesn't work for you.

Klipper just ignores any commands after failed init on start. That is the issue. At least it was at the moment I created the issue. I described it a few times, see above.

> And, individuals that know what they are doing, can modify the host Python code accordingly

But from that point these individuals are forced to maintain their own branch and merge new changes by their own.

> I'm curious, though, why enabling the power supply during startup doesn't work for you 

Because I don't want to enable main PSU on start.
Why?
It will remain powered on in case of short power outage(i.e reboot)
RPi is always on(via aux PSU) and main PSU should be turned a second before we need it.
Of course there's an option to power it off after initialization. But what for? 
We don't really need to start it at all.

>  I don't think it make sense to burden them with config documentation that effectively says "don't use this config option".

It effectively says just 'Be careful and initialize manually'
It's useful for me. I've been using it for a month. And I believe I'm not the only person who needs this.

cybertza, 2019-10-20:
Hi Guys
Also been using the Klipper, and i must admit i am loving it's functionality and ease of use, I am also having this same issue, so i am pitching in to see if i can help with ideas, or atleast get notifications when this is resolution.

My config is as Follows
Octoprint + Ender 3 + MKS1.4 + Rpi4 + Bltouch + Relay + cura direct print (still need to figure sd out for the progress to work) 
(still to do is MKS Mosfet - Will get around to it)

For Relay control i am using the RPI GPIO plugin (https://github.com/kantlivelong/OctoPrint-PSUControl) since that was organically how the project grew (mostly since that's where the relay lived on the housing i found on thingiverse :p)

So for me to do a print i need to do the following:

Go onto OctoPrint => click Power On
Go to klipper page => Click reboot Firmware.
And things are magical.

My standard errors range from ADC out of range or that my Stepper doesn't reply:
If I power up and I try the fix to init the stepper, it doesn't work, since the printer is halted. 
Send: INIT_TMC STEPPER=stepper_x
Recv: // Unable to read tmc uart 'stepper_x' register IFCNT
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Printer is halted
Recv: !! Unable to read tmc uart 'stepper_x' register IFCNT

So i know to resolve the issue 2 things has to happen,
1 x Power needs to be supplied.
2 x A Firmware reset needs to be done.

If i try to do this programmatically with GCODE before the print starts i get the "printer not available" during the start, and the print aborts.
Now i could be adventurous and change the Error handling to ignore under octoprint => serial => Behaviour. ( but that seems like a bad idea).

So my recommendation would be for Klipper to handle this more gracefully to encourage backward compatibility.

I know there is a "busy" (busy: processing) marker available that can be sent from the printer, so instead of Printer not ready being an error, can it be a info behind a wait_busy (for 5 cycles) then an error. That way it gives the firmware some time to restart\refresh everything needed until a bettor solution comes along.
So ergo the nice solution for me would be 1:
Either the M80 command get handled by the Host and does a configurable
1. Start PSU (wait 500ms) [M80]
2. Restart Firmware [FIRMWARE_RESTART] (wait 500ms)
3. Display Ready
4. Magic Happens.
So ergo the nice solution for me would be 2:
Or the "host software" handles the FIRMWARE_RESTART command better by reporting "wait/busy" when it knows its issued a FW restart (also can't hurt to put a M117 Restarting when you restart the host, and FW restarting for FW.

So ergo the nice solution for me would be 3:
A configurable "errors to recover from sections" for the host. 
Aka I know i am going to receive 1 of 2 errors, ADC or Stepper.
I just need those gracefully handled without breaking prints, aka
if the Host receives either of these messages tells octo to -busy wait- do the following:
1. Start PSU (wait 500ms) [M80]
2. Restart Firmware [FIRMWARE_RESTART] (wait 500ms)
(keep a count - Fail on 3rd attempt)

The aim is to have it as "easy as possible"
For me i want to click print in cura, and printer starts up, and makes magic.
but also to keep all the safety in place.
If there was a way to read some Vref on the board to tell you if its powered on would also help a lot with knowing why there is an error. and that could be displayed if feasible (Stepper failed - No Power on board) although not really important at all.) 

These are just ideas i have been having, if the PSU control plugin had more functionality like Execute Gcode after Power On it would also be just as great. That said, for someone with your skills it may be easy to add PSU control to the Klipper Page that does this all gracefully, People that tinker with this firmware are mostly going to have the TMC and Relays, and mosfets, and BLtouches. that to the current list of feature :-)

I will also migrate the Relay to the MKS if that's a workable solution.

Thank you so much for the time and effort that you put into this, and although initial configuration takes some time, once you start getting there and understand it, it becomes amazing.
for new users you may get a lot of joy from making an "config assist page"
where you basically have a wizard based website that spits out a cfg file.

cybertza, 2019-10-20:
Perhaps this could be changed to an Info+ Wait when a restart was called.
6 seconds will be a good spot.
https://github.com/KevinOConnor/klipper/blob/8c69e1bcfe2aa7f17fb919bf0131ee81d85c3976/klippy/gcode.py#L750
Although im not sure if onboard PSU control lives past a firmware restart?

KevinOConnor, 2021-02-21:
FYI, PR #3974 removes the error if a tmc driver can not be contacted during startup.  (Instead, a shtudown is raised if the driver can not be contacted when it is actually enabled.)

-Kevin

KevinOConnor, 2021-03-05:
This should be addressed now with PR #3974 .

-Kevin

closed 2021-03-05 by KevinOConnor