Klipper3d/klipper #1703: Failed to home z: Timeout during endstop homing

EvilGremlin, 2019-06-05:
I have a optical Z probe with servo, but it cannot home if it needs to travel more than 5-10mm, it just stops midway. XY homes just fine. In the log i homed Z from zero height, then raised it 20mm and then homing failed.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3256459/klippy.log)

KevinOConnor, 2019-06-06:
Best guess is that the stepper_z step_distance is not accurate - make sure to run the checks at https://www.klipper3d.org/Config_checks.html .  It's also possible the Z stepper is losing steps - you may want to double check its wiring.

-Kevin

EvilGremlin, 2019-06-06:
Nope, absolutely everything is fine. Marlin works, Klipper don't.

KevinOConnor, 2019-06-18:
Any further updates on this?

-Kevin

EvilGremlin, 2019-06-18:
Nope, it doesn't work with right settings, so i flashed back marlin for now.

KevinOConnor, 2019-06-28:
Okay.  Lets close this issue then.

-Kevin

closed 2019-06-28 by KevinOConnor