Klipper3d/klipper #1704: High MCU load when using SSD1306 display

modmt, 2019-06-05:
Hello all, 
on my 3D DIY printer I installed an SSD1306 OLED display. 
It's working fine, but I got lot of load (nearly 200% on Raspi 3B)  on MCU compared to my solution before without any display (load max. 35%).
Also when I use another display (RepRapDiscount 128x64 Full Graphic Smart Controller) i got only max. 40% load. Unfortunatelly the RepRap 128x64 display is to big and will not fit in my design of my (small) printer, so the SSD1306 display is perfect in this case.

Is this a know issue and could this be optimized to reduce the load? 
in the performance graph you can see it. Until 1:05 it's with the SSD1306, after that with RepRap display.
[image]

Thank you for your help.

modmt, 2019-06-05:
Here is the klippy.log
[klippy.log](https://github.com/KevinOConnor/klipper/files/3258647/klippy.log)

KevinOConnor, 2019-06-06:
All lcd updates use the lowest priority in Klipper - so basically, the lcd updates only occur if the micro-controller is otherwise idle.

That said, the 200% load certainly does look odd.  The load measurement is just a heuristic and it is possible something in the lcd updates are causing the heuristic to behave oddly.  I'll try to take a look if I get time.

-Kevin

modmt, 2019-06-06:
Thank you for your answer Kevin.
I’m calmed down if you mean that its only an failure in the heuristic and real load could be lower.
I got this high load displayed only with the SSD1306 display.

Anyway it would be nice if you could check this if you get time.
Thank you.

KevinOConnor, 2019-06-18:
FYI, I was unable to reproduce the high ssd1306 issue locally (though I'm not using the exact same hardware).  Your log didn't show high load either - though that's probably because it wasn't a log from a print.  Would you have a log from a session that does show the very high load?

-Kevin

KevinOConnor, 2019-07-10:
I'm going to close this issue as it looks like the conversation has concluded.  Let us know if you can obtain a log with the high load.

-Kevin

closed 2019-07-10 by KevinOConnor