Klipper3d/klipper #1705: Current Feedrate

smark-, 2019-06-05:
Hi all, I was wondering if there was a way to retrieve the current printer feedrate? I'd like to have a feedback on the current printing speed as it happens with the RepRap firmware

KevinOConnor, 2019-06-06:
It's possible to get at this information via a macro using the new g-code template system (see 
https://www.klipper3d.org/Command_Templates.html ).  It's via the undocumented `printer.gcode.speed` (in mm/min).

Otherwise, it is also available via writing a Python host "extras" module.

-Kevin

Hywelmartin, 2019-06-07:
For a "as and receive ones" sure but as a speedo I think that the amount of chit chat between octo and Klipper will get too much... Is there a way to bypass octoprint and get the information straight to some sort plugin in octoprint..

KevinOConnor, 2019-06-09:
@Hywelmartin - I'm not sure if you were asking me that question.  Currently, the only way to get info from Klipper to OctoPrint (plugin or other) is via OctoPrint sending a g-code command to Klipper.

-Kevin

Hywelmartin, 2019-06-10:
@KevinOConnor I more or less let it out in the wild.. @smark- is asking for a speedometer, like the one in the RRfw, and to even gain any useful information, it has to have rather high rate in the communication.. and I don't think that octoprint can handle that..

KevinOConnor, 2019-06-12:
I don't understand what's being reported here, or what's "in the wild".

-Kevin

closed 2019-06-28 by KevinOConnor