Klipper3d/klipper #1708: strange bed level behaviour

NicoSoftware, 2019-06-06:
Hey!
I'm having a strange problem with my saved meshed bed. When I ```BED_MESH_CALIBRATE``` it's perfectly calibrated and all values in my config are correct.
But when I start my print, it's always completly off. Most of the time the nozzle is too low, sometimes too high. Is that because of the heat?
Should I calibrate when my bed is heated up?
Klippy.log:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3262915/klippy.log)
printer.cfg:
[printer.txt](https://github.com/KevinOConnor/klipper/files/3262919/printer.txt)

drphil3d, 2019-06-07:
check to make sure you don't accidently have one of your axis reversed. easy mistake to make that will throw off everything.

NicoSoftware, 2019-06-07:
> check to make sure you don't accidently have one of your axis reversed. easy mistake to make that will throw off everything.

I will, thanks!
I've also noticed it going up about 2cm into the air at some point in the meshed bed, somethings really wrong here ..

drphil3d, 2019-06-07:
well, did you look at your height map? Your bed is pretty far off from level 4mm from one side to the other. It makes perfect sense to me whats going on. also your home override gcode looks like it could be causing issues. end with G28 so it's homed instead of G1 Z0. You may want to implement fade height.

NicoSoftware, 2019-06-08:
> also your home override gcode looks like it could be causing issues. end with G28 so it's homed instead of G1 Z0.
I don't really see the logic behind this.
Do you mean I should define the macro like this?
```
[homing_override]
axes: z
set_position_z:+15
gcode:
    G91
    G1 Z+10 F1000
    G90
    G28 X0 Y0
    G1 X53 Y18 F3200
    G28 Z0
    G1 Z5 F1000
    G1 X0 Y0 F3000
    G1 Z0 F1000
    G28
```
That would be a recursion haha
And why should it be defined different? It worked perfect until now to home all axes and without having trouble crashing into the bed..

> well, did you look at your height map? Your bed is pretty far off from level 4mm from one side to the other.
Yeah I did! But why is it shooting up so high? I mean it's 4mm and not 20mm ..

> You may want to implement fade height.
- That's helpful, I will look into this. thanks!
Is fade height under the ```[bed_mesh]``` section?
[image]

Arksine, 2019-06-08:
Leave fade disabled until you know the mesh is working.  When you are ready to enable it you set fade_end, 10 is a good starting point (the mesh fade will be complete by z=10).

I'm looking at your log and the saved mesh is showing a deviation of nearly 5mm.  Either something is wrong with your probing setup or your bed is seriously tilted.   You should also calibrate when the bed is at print temperature, heat can cause the geometry to change.

drphil3d, 2019-06-08:
hahah I didn't understand until just now what the purpose of fade height is really is. I thought it was to fade out measurements that where out of bounds not fade the leveling out. thats hilarious.

NicoSoftware, 2019-06-10:
> I'm looking at your log and the saved mesh is showing a deviation of nearly 5mm. Either something is wrong with your probing setup or your bed is seriously tilted. You should also calibrate when the bed is at print temperature, heat can cause the geometry to change.

Hey, 
Yes, I've noticed this too, but the probing runs smooth and perfectly, it just save's wrong points..
I've calibrated with a heated bed..

BlackStump, 2019-06-10:
Have you checked and calibrated your probe as I mention before your probe offset is wrong.
a normal bltouch offset is around 1.55 you have 0.0025
this is from your config-
[bltouch]
sensor_pin: ^PC4 ; P1.24
control_pin: PA4 ; P1.26
pin_up_reports_not_triggered: true
pin_up_touch_mode_reports_triggered: False
pin_move_time: 0.200
x_offset: -46
y_offset: -2
z_offset: 0.0025  <====this is wrong 
speed: 5
[image]

https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md

NicoSoftware, 2019-06-10:
> Have you checked and calibrated your probe as I mention before your probe offset is wrong.
> a normal bltouch offset is around 1.55 you have 0.0025
> this is from your config-

Hey blackstump,
Yeah I did calibrate it, I've also tried everything (0.2 steps) from ```z_offset: 0.0025``` to ```z_offset: 2.0```.
But it didn't work, some saved mesh points are just completely offtarget, but others are perfect fine ..

BlackStump, 2019-06-10:
if you followed the steps to calibrate your probe then I am not sure why you would use a number different from what the calibration told you the probe offset was.
something mechanically is wrong with your printer. It is a Ender3 right?

NicoSoftware, 2019-06-10:
> if you followed the steps to calibrate your probe then I am not sure why you would use a number different from what the calibration told you the probe offset was.

Yeah should be allright, I've measured it and all the values are given by the petsfang v2 repository on thingiverse anyway. They really should fit. But I'm calibrating my z_offset again. I really don't think that this will make the difference. The Nozzle goes up into the air 2cms, so I think the problem is somewhere else.
This is my meshed bed after calibrating again with a heated up bed:
[image]

Arksine, 2019-06-10:
Its still measuring over 4mm of tilt in your bed along with periodic spikes in the probe's measurement.  The spikes indicate there is something wrong with the probe.

If you truly have ~4mm on tilt on your bed I strongly recommend using mechanical means to level it as much as possible prior to any kind of software correction.

Also I don't recommend using lagrange interpolation when probing a 7x7 grid, switch to bicubic.

NicoSoftware, 2019-06-10:
> If you truly have ~4mm on tilt on your bed I strongly recommend using mechanical means to level it as much as possible prior to any kind of software correction.

No, the values reported by the bltouch aren't correct. While printing my nozzles is flying in the air at some points.

> Also I don't recommend using lagrange interpolation when probing a 7x7 grid, switch to bicubic.

I will look into it, thanks!

BlackStump, 2019-06-11:
suggest run BED_MESH_CLEAR
then run the following instructions to check your probe repeatability 
https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md#repeatability-check
do the test above your 4  bed adjusting screws on your ender3 and post the results

NicoSoftware, 2019-06-11:
> suggest run BED_MESH_CLEAR
> then run the following instructions to check your probe repeatability
> https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md#repeatability-check
> do the test above your 4 bed adjusting screws on your ender3 and post the results

I'll! thank you:)
Hopefully I have time after working to test my probe!

KevinOConnor, 2019-06-18:
Any further updates on this?  If you're getting a consistently low nozzle height, then one should be able to address that by following the directions at: https://www.klipper3d.org/Bed_Level.html

-Kevin

NicoSoftware, 2019-06-20:
> Any further updates on this? If you're getting a consistently low nozzle height, then one should be able to address that by following the directions at: https://www.klipper3d.org/Bed_Level.html
> 
> -Kevin

Hey Kevin, 
Thanks for the help, but I got it working by increasing the samples.
It's actually my BLtouch reporting inaccurate values at some probes.
Closing this! :)

closed 2019-06-20 by NicoSoftware