Klipper3d/klipper #1711: Wanhao I3 plus Mark 2 z Probe

nzlrhyz, 2019-06-07:
Using information found here https://sebastien.andrivet.com/en/posts/wanhao-duplicator-i3-plus-3d-printer/ and the default Plus config I managed to get it all working up until the Verify endstops in config check - no matter what Z is always open, I've tried both PA1 and PA3 including !, ^, ~ and multipleo f them 

# This file contains pin mappings for the Wanhao Duplicator i3 Plus
# (circa 2017).  To use this config, the firmware should be compiled
# for the AVR atmega2560.
# Pin numbers and other parameters were extracted from the
# official Marlin source available at:
# https://github.com/garychen99/Duplicator-i3-plus

# See the example.cfg file for a description of available parameters.

[probe]
pin: PA3
z_offset:2

[stepper_x]
step_pin: PF7
dir_pin: !PK0
enable_pin: !PF6
step_distance: .0125
endstop_pin: ^!PF0
position_endstop: 0
position_max: 200
homing_speed: 30.0

[stepper_y]
step_pin: PK2
dir_pin: !PK3
enable_pin: !PK1
step_distance: .0125
endstop_pin: ^!PA2
position_endstop: 0
position_max: 200
homing_speed: 30.0

[stepper_z]
step_pin: PK5
dir_pin: PK7
enable_pin: !PK4
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
position_max: 180
position_min: 1

[extruder]
step_pin: PF4
dir_pin: PF5
enable_pin: !PF3
step_distance: 0.010417
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PG5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PF1
control: pid
pid_Kp: 30.850721
pid_Ki: .208175
pid_Kd: 192.298728
min_temp: 0
max_temp: 260

[heater_bed]
heater_pin: PE5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PK6
control: pid
pid_Kp: 64.095903
pid_Ki: 1.649830
pid_Kd: 622.531455
min_temp: 0
max_temp: 110

[fan]
pin: PE3

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 800
max_z_velocity: 5
max_z_accel: 100

[bed_mesh]
min_point: 20,20
max_point: 200,200
probe_count: 4,4

[klippy.log](https://github.com/KevinOConnor/klipper/files/3265647/klippy.log)

KevinOConnor, 2019-06-09:
What type of probe does the printer have?

If you have a probe, you would typically verify it using QUERY_PROBE (and not QUERY_ENDSTOP).  However, if the probe is a bltouch, it has a different test sequence (see https://www.klipper3d.org/BLTouch.html ).

-Kevin

nzlrhyz, 2019-06-09:
I figured it out, the probe was on another pin that was different from the original (Same with the Y enable), I did make a PR with the config I used for the Mark2

KevinOConnor, 2019-06-18:
I'm going to close this issue as it looks like a solution was found for the original issue.

-Kevin

closed 2019-06-18 by KevinOConnor