Klipper3d/klipper #1714: Intersecting Infill over extrusion causing nozzle to drag with higher density infill ... Pressure Advance?

drphil3d, 2019-06-09:
I'm having an issue on multiple printers running the latest code where when printing a higher density triangular infill pattern 30% or higher there seems to be an issue with over extrusion where infill overlaps itself. I've adjusted my extruder steps, filament width and pressure advance but can't seem to fix this issue. 

Maybe we need a better model for tuning pressure advance. It's just really hard to tell if it's set right. Would something similar to Marlins linear advance test generator work?
http://marlinfw.org/docs/features/lin_advance.html

[klippy.log](https://github.com/KevinOConnor/klipper/files/3269029/klippy.log)

KevinOConnor, 2019-06-09:
If you've calibrated the extruder steps, and are then seeing overlap, it sounds like a slicer issue.  Slic3r is really bad at that, for example.

As for tuning pressure advance, in my own testing I've had poor results with the Marlin method - slight changes in bed level can completely skew the results.  There has been some interest in changing to a "pressure advance tower" method - see PR #1243 for further info.

-Kevin

Hywelmartin, 2019-06-11:
The crossings in that infill is 3 lines stacked ontop of each other/ layer 
you can set the infill line width in Print Settings/Advanced but I don't think that the result will be different 

[image]

drphil3d, 2019-06-11:
I think I found the perfect model for PA tuning 
https://www.thingiverse.com/thing:2371069

drphil3d, 2019-06-12:
@KevinOConnor I found this pressure advance calibration model and python script on thingiverse
https://www.thingiverse.com/thing:3442071/files

theopensourcerer, 2019-06-13:
@drphil3d This one worked really well for me:

https://github.com/PrusaOwners/prusaowners/wiki/Pressure_Advance

closed 2019-07-21 by KevinOConnor