Klipper3d/klipper #1716: Error doing a PID tune

PerLangkjaer, 2019-06-09:
Good morning all

Yesterday i changed from a V5 to a V6 all-metal hotend and wanted to do a new PID tune.
Now i got this message:

Recv: // Heater extruder not heating at expected rate
Recv: // See the 'verify_heater' section in config/example-extras.cfg
Recv: // for the parameters that control this check.
Recv: //

And as you can see it aborts in the 2nd cycle at 191-192 degres Celsius.
I have solid readings from the thermistor I think and the heater seems to heat up good.
Its the same heatblock, heater and thermistor i use.

[image]
[klippy.log](https://github.com/KevinOConnor/klipper/files/3269188/klippy.log)

KevinOConnor, 2019-06-09:
You should upgrade to the latest version of Klipper - there has been several improvements to the verify_heater mechanism.

-Kevin

KevinOConnor, 2019-06-24:
I'm going to close this issue as it looks like the original question was answered (upgrade the software).

-Kevin

closed 2019-06-24 by KevinOConnor

PerLangkjaer, 2019-06-25:
Hello and sorry Kevin
For not having updated at the matter.
It where not a straight forward update, it took me 2 evenings and a reinstall to solve the update thing.

It did not do the trick with an update, so I took the hotend apart again and applied a little thermal paste to the thermistor and that surely solved my problem.

I just need to give you a shout out for your amazing work :-)

Regards 
Per