Klipper3d/klipper #1720: Arduino DUE shutdown when run FORCE_MOVE

semos2k, 2019-06-09:
I installed the latest version of klipper on an arduino DUE, when I run FORCE_MOVE klipper is shutdown
[image]
 
[klippy-due.txt](https://github.com/KevinOConnor/klipper/files/3269699/klippy-due.txt)

KevinOConnor, 2019-06-10:
The FORCE_MOVE command is a diagnostic and debugging command.  If you attempt to move at a high velocity without using acceleration (or use an acceleration higher than the printer was originally configured for) then it can cause the micro-controller to flag those moves as errors.  It's not something we plan to fix as we don't recommend using FORCE_MOVE for anything other than debugging purposes.

-Kevin

semos2k, 2019-06-10:
hello kevin, I tried to make a home and I obtained the same result.

KevinOConnor, 2019-06-12:
You'd need to attach the Klipper log file from that event in order for us to assist.

-Kevin

closed 2019-06-24 by KevinOConnor