Klipper3d/klipper #1722: BLTouch probe is raised / Error: BLTouch failed to raise probe

Ladrogue, 2019-06-10:
dear all, i am using klipper with a original bltouch since few days now... i sometimes (very often) getting the following errors:

Send: G29
Recv: // probe at 78.000,30.000 is z=1.875000
Recv: // probe at 78.000,30.000 is z=1.877500
Recv: // probe at 154.000,30.000 is z=2.580000
Recv: // probe at 154.000,30.000 is z=2.582500
Recv: // probe at 230.000,30.000 is z=3.220000
Recv: // probe at 230.000,30.000 is z=3.220000
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: !! BLTouch failed to raise probe
Changing monitoring state from "Operational" to "Error: BLTouch failed to raise probe"
Changing monitoring state from "Error: BLTouch failed to raise probe" to "Offline (Error: BLTouch failed to raise probe)"
Connection closed, closing down monitor

can someone tell me whats wrong? i already asked google for hours but still no solution :(

Ladrogue, 2019-06-10:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3272261/klippy.log)

KevinOConnor, 2019-06-12:
If you have an original bltouch, it indicates a wiring or bltouch failure.  Perhaps extruder movement is pulling on one of the wires.  If you have a clone, it may require `pin_up_reports_not_triggered: False` in the config.

-Kevin

closed 2019-06-28 by KevinOConnor