Klipper3d/klipper #1723: Klipper stops in middle of print

VanKurt, 2019-06-10:
Currently I can't get any print longer than an hour to complete with Klipper.
Today my printer stopped printing after two hours (and the hot nozzle kept burning into the printed object for another hour).

In the Octoprint log I found the following:
> 2019-06-10 18:08:51,035 - octoprint.util.comm - WARNING - Received an error from the printer's firmware: Lost communication with MCU 'mcu'
> | Last lines in terminal:
> | Send: N119686 M205 X30 Y30*20
> | Recv: // Unknown command:"M205"
> | Recv: ok
> | Send: N119687 G0 F7200 X125.048 Y102.175*113
> | Recv: ok
> | Send: N119688 M204 S500*92
> | Recv: ok
> | Send: N119689 M205 X20 Y20*27
> | Recv: // Unknown command:"M205"
> | Recv: ok
> | Send: N119690 G1 F4800 X127.548 Y102.175 E366.48635*60
> | Recv: // Klipper state: Shutdown
> | Recv: ok
> | Send: N119691 G1 X127.548 Y100.875 E366.50797*92
> | Recv: // Lost communication with MCU 'mcu'
> | Recv: // Once the underlying issue is corrected, use the
> | Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
> | Recv: // config, and restart the host software.
> | Recv: // Printer is shutdown
> | Recv: !! Lost communication with MCU 'mcu'

I'm not sure what caused the problem. There seem to be plenty (unknown/evil?) M205 commands. Also a weird "N119690" just before the shutdown.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3273458/klippy.log)
[octoprint.log](https://github.com/KevinOConnor/klipper/files/3273459/octoprint.log)

theopensourcerer, 2019-06-10:
I do not believe that Klipper supports M205 - you probably need to change the "Printer type" in your slicer. Recommendations for this are in the documentation.

https://reprap.org/wiki/G-code#M205:_Advanced_settings

Unfortunately your klippy.log doesn't contain any of the previous printing information before a restart. It seems like the MCU has "gone away" as Klipper cannot connect to it at all.

mcmatrix, 2019-06-11:
Yep, Klipper is not supporting M205.
After unknown command, it goes into shutdown state.
The weird `N119690` is a line number for next GCode. 
Klipper internally is just ignoring it.
You have to check your slicer as @theopensourcerer said.
If you cannot get rid of this in slicer then another option is to make a dummy
GCode macro for M205
```ini
[gcode_macro M205]
gcode:
```

theopensourcerer, 2019-06-11:
I see you use TMC2208s - you might want to play around with the Stealthchop threshold, or try disabling Stealthchop completely. See the Faq and config examples for more info.

VanKurt, 2019-06-11:
> I see you use TMC2208s - you might want to play around with the Stealthchop threshold, or try disabling Stealthchop completely. See the Faq and config examples for more info.

I set the threshold to a very high value so that the TMC can never leave stelthchop mode. In spread cycle there is a super loud grinding noise, which sounds as if the whole printer would explode ;-)
In stealth chop it runs very smooth and quiet,
Is that a problem? Can't Klipper run the drivers in stealthchop reliably?

theopensourcerer, 2019-06-11:
Please see the FAQ and you might also want to review, the now closed, ticket #196 to get more information.

https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#my-tmc-motor-driver-turns-off-in-the-middle-of-a-print

NB: I'm not saying this is your problem. I am saying there have been problems in the past with 2208s. Without a Klippy.log of the actual print and the point at which it fails, it will be difficult to come to an accurate conclusion I should think.

VanKurt, 2019-06-11:
Thanks for your input. Today I redid the test print and saved the logs immediately after the print failed. This time it took only a few minutes until the problem occurred.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3277332/klippy.log)
[octoprint.log](https://github.com/KevinOConnor/klipper/files/3277336/octoprint.log)

jakep82, 2019-06-11:
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#i-keep-getting-random-lost-communication-with-mcu-errors

KevinOConnor, 2019-06-28:
I'm going to close this issue - as mentioned, the best advice we have on "lost communication with mcu errors" is in the FAQ.

-Kevin

closed 2019-06-28 by KevinOConnor