Klipper3d/klipper #1724: Printing with raft crashes MCU

brazda, 2019-06-11:
Everything works fine until I try to print with a raft. Anytime I use a raft my Z MCU crashes.

 MCU 'z' shutdown: Rescheduled timer in the past
This generally occurs when the micro-controller has been
requested to step at a rate higher than it is capable of
obtaining.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

octoprint.util.comm - WARNING - Received an error from the printer's firmware: MCU 'z' shutdown: Rescheduled timer in the past
| Last lines in terminal:
| Recv: // This generally occurs when the micro-controller has been
| Recv: // requested to step at a rate higher than it is capable of
| Recv: // obtaining.
| Recv: // Once the underlying issue is corrected, use the
| Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
| Recv: // config, and restart the host software.
| Recv: // Printer is shutdown
| Recv: !! MCU 'z' shutdown: Rescheduled timer in the past
| WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: MCU 'z' shutdown: Rescheduled timer in the past
| Recv: ok
| Send: N724 G0 F7200 X315.2 Y127.11*123
| Recv: // MCU 'z' shutdown: Rescheduled timer in the past
| Recv: // This generally occurs when the micro-controller has been
| Recv: // requested to step at a rate higher than it is capable of
| Recv: // obtaining.
| Recv: // Once the underlying issue is corrected, use the
| Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
| Recv: // config, and restart the host software.
| Recv: // Printer is shutdown
| Recv: !! MCU 'z' shutdown: Rescheduled timer in the past
[octoprint (1).log](https://github.com/KevinOConnor/klipper/files/3276685/octoprint.1.log)

jakep82, 2019-06-11:
You've posted the log from Octoprint which is not helpful. You need to post your Klipper log.

brazda, 2019-06-11:
oops I guess I hit submit before if uploaded fully. I will post the log in 24 hours as I rebooted and no longer have the log so I will need to recreate the failure.

brazda, 2019-06-12:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3282452/klippy.log)

jakep82, 2019-06-12:
The error in your first post appears to be correct.  You need to reduce your max velocity on the Z axis to around 10mm/s, or reduce the microsteps you're using on your Z steppers.  At 50mm/s you're pushing well beyond the limits of a 2560.

brazda, 2019-06-13:
ok that fixed it thanks. Wonder why that has not came up before. Been using it for awhile just never needed a raft before

closed 2019-06-13 by brazda