Klipper3d/klipper #1727: Support for half-stepping rotary encoders

modmt, 2019-06-13:
Hello all,

I'm using an SSD1306 display and Ky 044 rotary encoder. Unfortunatelly this encoder needs two steps to go one step further in the menu of the display.

I know that there is a thread with this topic, i did a comment but it is closed so I don't know if anybody else will read it... (https://github.com/KevinOConnor/klipper/pull/730)

Unfortunatelly that modification doesn't work for me. I always get this error:
[image]

I modified the menu an buttons.py like in the thread. Here you can find the files (i renamed it to .txt but with .py there is no upload possible).
I also added the encoder_steps_per_detent: 4 in the display section of the 

cfg.

[menu - Kopie.txt](https://github.com/KevinOConnor/klipper/files/3285858/menu.-.Kopie.txt)
[buttons - Kopie.txt](https://github.com/KevinOConnor/klipper/files/3285859/buttons.-.Kopie.txt)

Did anybody try this modification with success?
Thank you 

[klippy.log](https://github.com/KevinOConnor/klipper/files/3285907/klippy.log)

FHeilmann, 2019-06-13:
try doing the indentation of your modifications with spaces instead of tabs and see if that fixes it. Klipper is complaining about a formatting error in the python file.

modmt, 2019-06-13:
Thank you for your help Florian. That was the exactly problem. I mixed tabs and spaces an so I got the problem. I'm new in python, so I think this was a typical error by a newbie ;-)

Now the half-stepping mode works perfect, so I can close this.
Thanks again.

closed 2019-06-13 by modmt