Klipper3d/klipper #1728: MCU 'mcu' shutdown: ADC out of range on Extruder thermistor after upgrading heatbed

robanderson, 2019-06-13:
So I have a MP Select Mini that came with an underpowered head bed heater.

Everything was working fine, I got a new 120W heater and wired it up through a external BIQU Heat bed expansion module to limited the additional 5A going through my Ramps board (previous 60W heater upgraded to 120W @ 12V is additional 5A current).  https://www.amazon.com/gp/product/B01HEQVQAK/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

[image]

Everything looked good so I did a heater bed PID tune at 60 degrees.  Saved the config, then did another PID Tune to 80 degrees.  Everything looked good.  Looking to print ABS I took the heatbed up to 100 degrees and as soon as the heat bed gets up to 100 degrees the Hotend temperature starts falling, goes into negative and causes a Klipper Shutdown event.

I've not touched the hotend at all, which is a genuine E3D V6 hotend and thermistor.  I've tested taking the hotend up to 185 degrees before heating the heat bed and if I do that Klipper does not shutdown the MCU.  I've attached another photo showing that the recorded hotend temperature becomes unstable when the heat bed is up to 100 degrees and is cycling a heat phase to maintain the 100 degree target temperature.  
[image]

BlackStump, 2019-06-14:
check you hotend thermistor wire, you may have damaged it when you replace the bed.

KevinOConnor, 2019-06-18:
The symptoms appear to be hardware related - for example, damaged thermistor wire, voltage instability, or ramps overheating.

-Kevin

closed 2019-07-10 by KevinOConnor