Klipper3d/klipper #1729: Question - Adafruit Grand Central M4

jalanjarosz, 2019-06-14:
1) Is it possible to use the pre-installed UF2 bootloader on the Adafruit SAMD51 boards with Klipper: https://learn.adafruit.com/adafruit-metro-m0-express-designed-for-circuitpython/uf2-bootloader-details ?  If so, what is the process to load the Klipper FW on or is it a requirement to flash the bootloader along with Klipper FW?

2) If bootloader + Klipper FW is the only way, are there instructions/recommendations on what settings to use when configuring the board with the `make menuconfig` command?

Thanks!

FHeilmann, 2019-06-14:
configure the bootloader size during the `make menuconfig` step to a size of 16KiB and make sure to set the correct MCU variant for the grand central (SAMD51P19A).
The uf2 bootloader also accepts regular flashing via bossac, so you can use the method described in the klipper docs to flash the board.

KevinOConnor, 2019-06-28:
I'm going to close this issue as it looks like the original question has been answered.

-Kevin

closed 2019-06-28 by KevinOConnor