Klipper3d/klipper #1732: BLTouch homing fails if endstop for stepper_z1 is defined

mariusoei, 2019-06-15:
Hi everyone,

I'm trying to switch from Marlin to Klipper and I came across a small problem.
I used the Anycubic i3 Mega demo config (I am using the small i3).

I also have a BLTouch sensor installed. Now the commands and everything work fine.
When I tried to home, the z axis would stop moving when the pin touched the bed, the BLTouch would retract and then be deployed again right away. The movement of the axis stops, but nothing else happens until the homing times out.

After a while I double checked the config for the z endstops and voilà: There was another endstop defined for stepper_z1 in addition to the virtual endstop that I defined for stepper_z.
So I assume that the steppers are stopped because one endstop is triggered, but homing isn't successful until the second endstop triggers as well.

I think using separate endstops for multiple z axis steppers should either be forbidden or each axis should keep moving until it hits is respective endstop (which would cause problems on nearly every machine with 2 z-steppers).

As it is right now, the steppers both stop moving (good), but homing isn't completed until the second endstop is triggered (how could it, the stepper isn't even moving).

Anyway, I solved the problem for me by just removing that second endstop (doesn't exist anyway, but it was in the example config).
I just wanted to let you know so either people would find this issue when they run into the same problem or you can find a solution for this.

Best,
Marius

[klippy.log](https://github.com/KevinOConnor/klipper/files/3293113/klippy.log)

KevinOConnor, 2019-06-18:
When using multiple Z endstops, each Z stepper does move until it hits its respective endstop.  It doesn't make sense to use multiple Z endstops with a "virtual probe endstop".  Klipper did not do a good job of error reporting here.

-Kevin

mariusoei, 2019-06-18:
Not sure if that's due to the use of the virtual probe endstop, but in my case both steppers stopped moving when the probe was activated.
You're right, it doesn't make sense. I figure this would be a simple check when reading the config. I would like to implement that and create a PR, but I haven't really understood the overall code yet :-)

Best,
Marius

KevinOConnor, 2019-07-10:
FYI, there were several recent changes to the homing code.  Is the Klipper error reporting still bad with the latest code?

-Kevin

mariusoei, 2019-07-10:
I'll try it out as soon as I get the time :)
Thanks for the update!

mariusoei, 2019-07-25:
I just tried it with the exact same config as before. I didn't see any kind of warning in the Octoprint terminal or in the klippy.log.
When homing, the behaviour is the same as before also. The only error I get is: 

Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: !! Failed to home z1: Timeout during endstop homing

So I guess if you know what you're looking for you can notice the mention of axis **z1**.

KevinOConnor, 2019-07-26:
Could you attach the log from that session?

Thanks,
-Kevin

mariusoei, 2019-07-26:
There you go... the first error is on line 343:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3436856/klippy.log)

KevinOConnor, 2019-07-27:
Okay, thanks.  At least Klipper is now logging the Z1 error message (even if it is a bit cryptic).

-Kevin

KevinOConnor, 2019-08-17:
Thanks.  I'm going to close this issue as it looks like the original misconfiguration is now getting logged (even if it is still a bit cryptic).

-Kevin

closed 2019-08-17 by KevinOConnor