Klipper3d/klipper #1734: OverflowError: can't convert negative number to unsigned

pitucorto, 2019-06-15:
Hello,

I have updated to the last version of Klipper with following commands:

cd ~/klipper
git pull
~/klipper/scripts/install-octopi.sh

After this update, my printer is not able to connect correctly to my boards.

Any idea of what are happening?

Thank you.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3293561/klippy.log)

BlackStump, 2019-06-15:
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-upgrade-to-the-latest-software

pitucorto, 2019-06-15:
I have followed the guide and updated both mcu.

The error persist still.

FHeilmann, 2019-06-17:
Looking at your log(and config) the first thing that catches my eye is that you did not specify which SPI bus is supposed to be used for your TMC2130 drivers (spi_bus parameter in your tmc2130 blocks)

KevinOConnor, 2019-06-18:
Your log indicates there is code you are using (an "autolevel" module) which isn't in the main Klipper repo.  You'll need to retry with the pristine Klipper code and post a log from that in order for us to assist you.

-Kevin

pitucorto, 2019-06-25:
Hello,

Sorry for my late feedback, I was out of my house.
The problem seem to be in SPI communication with TMC2130 drivers. When I disabled them, everything was fine.

I have updated klipper to the last est commit and the problem seem to be fixed.

Thank you for your support.

pitucorto, 2019-06-25:
Hello,

This issue come back after do a "firmware_restart". For resolve it again, I executed following command in ssh:

```
~/klipper/scripts/install-octopi.sh
make menuconfig
make clean
make
sudo service klipper stop
make flash FLASH_DEVICE=/dev/ttyACM0
sudo service klipper start
[klippy.log](https://github.com/KevinOConnor/klipper/files/3326665/klippy.log)
```

However, after a "firmware_restart" the issue come back. Attached log file with these tests.

pitucorto, 2019-06-25:
I have two board with arduino due. 

When I disable TMC2130 from [mcu zboard] and do a "firmware_Restart" the issue is resolved.

I have tried to interchange both boards and the result is the same so I think that the hardware is OK (also the problems were when I updated to latest commit).

Attached klipper log.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3326767/klippy.log)

KevinOConnor, 2019-06-25:
This appears to be a defect in Klipper (it's not valid to use a print_time=0.0).  I'll put together a fix.

-Kevin

KevinOConnor, 2019-06-25:
Hopefully this is fixed now (commit 6ec7dee0).

-Kevin

pitucorto, 2019-06-26:
Thank you Kevin,

Everything is ok now.

Regards

closed 2019-06-30 by pitucorto