Klipper3d/klipper #1735: Babystepping Issue

velothump, 2019-06-16:
Babystepping with SET_GCODE_OFFSET doesn't seem to be working as expected. In https://github.com/KevinOConnor/klipper/issues/306 there's mention of a fix to the code to apply the changes on the next gcode command instead it seems to be applying (in the case of an Z adjustment) on the layer change. This was tested and repeatable by running a SET_GCODE_OFFSET Z=10, and it is applied at the layer change. This is running the HEAD of master.

Arksine, 2019-06-16:
The default behavior SET_GCODE_OFFSET was recently changed to apply the offset on the next Z move. This was done to prevent an error in the event than an offset was applied to a retraction.

You can get it to move immediately using the MOVE=1 parameter, ie:
```
SET_GCODE_OFFSET Z=10 MOVE=1
```

floppyslotter, 2019-06-16:
I hate to hijack a thread but I am having the same exact issue and didn't want to post back to back coordinate offset issues.  After waiting over a year between klipper firmware updates, I noticed the coordinate offset feature just straight up does not function, at all, at least from what I can tell.  I like to use it like the marlin babystepping feature when laying down the first layer to get the z-offset dialed in on the fly; I use one of those flex plate build surfaces so the offset may change a few thousandths between print jobs when I'm flexing the plate to remove parts.  But now immediately after updating klipper I noticed that I can no longer do that.  From Arksine's comment I gather that the offset move won't apply until the next change in z height, IE. the next layer?  Because I've tried adjusting the z offset up and down by over 1mm and it doesn't affect anything even on the next layer of printing when it theoretically should be printing spaghetti into the air or crashing into the bed.

Do I need to apply the SET_GCODE_OFFSET MOVE=1 command manually in the terminal to get it force a change?  Is there a way to add it to the existing offset?

Here is my log:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3293839/klippy.log)

velothump, 2019-06-16:
Thanks @Arksine , completely missed that in the G-Code docs.

closed 2019-06-16 by velothump

Arksine, 2019-06-16:
@floppyslotter The default behavior is to apply it on the next layer.   Adding MOVE=1 to the command offset will apply it immediately:
```
SET_GCODE_OFFSET Z=1 MOVE=1
```
So for example, if you are using the default menu, you could override the menu item to enable baby stepping:
```
[menu __tune __offsetz]
type: input
name: "Offset Z:{0:05.3f} "
parameter: gcode.homing_zpos
input_min: -5
input_max: 5
input_step: 0.005
realtime: true
gcode: SET_GCODE_OFFSET Z={0:.3f} MOVE=1
```