Klipper3d/klipper #1737: Testing STM32F1/F4 when home only X, Y motor move

semos2k, 2019-06-16:
Hi, I am triying klipper on STM32F1 (SKR Mini) ans STM32F4 (STM32F407vet6) when execute home on X motor, Y motor move en same direction and stop when X endStop is touched.

I test move motor independently with FORCE_MOVE and work correctly.

This file use skr mini and anet board only for test lcd.

[klippy-stm32.log](https://github.com/KevinOConnor/klipper/files/3294369/klippy-stm32.log)

on stm32f407 i used this pull request for test, same issue. https://github.com/KevinOConnor/klipper/pull/1382.

semos2k, 2019-06-16:
I test only SKR Mini (without Anet board) and go to home X, same result.

semos2k, 2019-06-16:
log file:
[klipper-stm32-no-anet.log](https://github.com/KevinOConnor/klipper/files/3294383/klipper-stm32-no-anet.log)

FHeilmann, 2019-06-17:
is your printer a corexy printer? Because on those kinematics both motors have to move to execute a pure X or Y move.

semos2k, 2019-06-17:
You're right, when I start the configuration forget to change the parameter for the type of printer, 

Thank you.

closed 2019-06-17 by semos2k