Klipper3d/klipper #1739: BLTouch just won't give up!

ozgunawesome, 2019-06-16:
This started happening recently, I don't know what exactly could be causing it.

I use BLTouch V3 on a SKR V1.3 board with bed_mesh calibration.

Normally it probes the 9 points with G29 at start, then shuts off and the print goes normally.

Recently the following is happening:
* At probe point 7 (upper left corner) the probe power cycles (I can tell by it going through the two stow/deploy cycles) but if the Z is high enough it still has time to work.
* I get a correct calibration and the print starts
* From there on forward, the BLTouch just randomly starts going on and off at random intervals, deploying and dragging the probe over the printed material causing all kinds of mess.
* I pause the print, unplug the BLTouch cable and resume, then the print goes fine.

What could be the cause of this? Has there been any change recently regarding the BLTouch code? Or is it possible that I'm drawing too much power from the power supply and it's causing voltage drops and the probe resets. (If yes why at this specific point?)

Attaching log file but it doesn't say anything.. Thanks for the help
[klippy.log](https://github.com/KevinOConnor/klipper/files/3294579/klippy.log)

jakep82, 2019-06-16:
It sounds like you have an intermittent issue with the power connection to the BLTouch.

ozgunawesome, 2019-06-16:
It's possible. I switched to bed tilt and gave a different pattern but it always gives up when it reaches the top Y point. But I haven't changed anything regarding wiring

ozgunawesome, 2019-06-16:
I also noticed that my printer refuses to move past XY 200 mm even though I set position_max: 235 for both.

ozgunawesome, 2019-06-16:
Anyway intermittent power issue confirmed thanks :)

closed 2019-06-16 by ozgunawesome