Klipper3d/klipper #1740: Many random errors

claustro, 2019-06-16:
I have many random errors. I double checked wiring and psu and everyting works ok . Also I don't have problem running MK4duo firmware.
The errors I am having are:

1) first I had 
Heater extruder not heating at expected rate but it seems to me , looking to the graph that the allarm was false

2) the I tryed to tweak [verify_heater extruder]
max_error: 400
check_gain_time:35
 An after 2 successfull print I started to receive at the start to the print 
Internal error on command:"G28"
MCU 'mcu' shutdown: ADC out of range

the errors happen at a specific time. 
Printer start to heating bed , then extruder, make 0 on x and when the printer start to zero Y the temperature on the lcd is switched from 235 ( what is programmed ) to 248 (?) and then the MCU error

After the error bed and nozzle remain on heating state.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3294703/klippy.log)

FHeilmann, 2019-06-17:
are you 100% sure that you have chosen the right thermistor type? E3D for example use a Semitec 104GT-2 on their hotends.

KevinOConnor, 2019-06-18:
The log indicates Klipper responded as designed - the reported extruder temperature jumped from 235 degrees to 255 degrees within a second.  That's clearly a failure of the temperature sensor and Klipper took the correct action by disabling the heaters.  I suspect the cause is a voltage issue, wiring issue, or similar.

-Kevin

claustro, 2019-06-18:
Thank for answering me.
I checked the thermistor and it is correct .
What is strange to me is that the temperature is set to 255 . In the lcd I can read 250->235 at the moment of the error. 
I also tried a brand new thermistor and heater cartridge with same problems.
I am gonna try a new PSU
It is very strange that the error happens always at about the same time .
Any idea what I could check?

KevinOConnor, 2019-06-18:
The log indicates the dramatic temperature change occurs after the printer starts homing.  Check that the thermistor wires aren't being crimped by carriage movement.  Check that the power supply isn't being overloaded.

-Kevin

closed 2019-07-10 by KevinOConnor