Klipper3d/klipper #1741: OctoPrint "Configuration" tab not updated after SAVE_CONFIG

joshpainter, 2019-06-18:
After a BED_MESH_CALIBRATE or a PROBE_CALIBRATE or a PID_CALIBRATE and then a SAVE_CONFIG, the Klipper Configuration tab in the Octoprint plugin is not refreshed with the latest changes. If you restart OctoPrint, the new changes show up. I expected to see the latest version of that file in that tab after a SAVE_CONFIG but it appears that it might be cached? This caused confusion because I thought SAVE_CONFIG was not actually saving anything even though it was. Log attached because you asked so nice but dunno if it helps anything. :)

[klippy.log](https://github.com/KevinOConnor/klipper/files/3303809/klippy.log)

theopensourcerer, 2019-06-19:
I would guess you should report this to the Octoklipper plugin not here:

https://github.com/mmone/OctoprintKlipperPlugin

Hywelmartin, 2019-06-19:
A refresh of the page normally sorts things out..

mental405, 2019-06-19:
> I would guess you should report this to the Octoklipper plugin not here:
> 
> https://github.com/mmone/OctoprintKlipperPlugin

Good luck with that. I don't think anyone has touched OctoKlipper in almost a year.

theopensourcerer, 2019-06-20:
Correct - but there are some more active forks ;-)

KevinOConnor, 2019-06-28:
FYI, as indicated, this is something that should be reported to OctoKlipper.

-Kevin

closed 2019-07-10 by KevinOConnor