Klipper3d/klipper #1742: Z- offset calibration

Profanum, 2019-06-19:
Z- offset doesnt change within the layer, it changes between the layers. It used to hop up and down within the same layer almost in real-time as i changed it with :
SET_GCODE_OFFSET Z_ADJUST=0.5
now it just changes between the layers. That makes it very hard to get just right adhesion to build plate because as the layer starts its already too late to adjust anyting, is there a way to solve this?

Profanum, 2019-06-19:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3305793/klippy.log)

Hywelmartin, 2019-06-19:
`SET_GCODE_OFFSET Z_ADJUST=0.5 MOVE=1`
If "MOVE=1" is specified then a toolhead move will be issued to apply the given offset (otherwise the offset will take effect on the next absolute G-Code move that specifies the given axis).

Profanum, 2019-06-20:
Both of solutions worked, Thank You very much, You made me a day :)

closed 2019-06-20 by Profanum