Klipper3d/klipper #1743: No module named jinja2 after update

supernielsen, 2019-06-19:
I have the exact same issue as #1501 
But hi solution does not solve my problem.

it says no module named jinja2 after update i tried everything i can think of.

mental405, 2019-06-19:
Have you updated to the latest version of OctoPrint?

supernielsen, 2019-06-20:
My octoprint is completely updatad.. Im seriously worried about losing all my settings so im not prepared to do it all over yet.. :)

theopensourcerer, 2019-06-20:
Have you tried:

https://github.com/KevinOConnor/klipper/blob/master/scripts/klippy-requirements.txt

supernielsen, 2019-06-20:
I will try that when i get home today and report back, thank you very much
for your response

FHeilmann, 2019-06-20:
Simply rerun the klipper installer to fix this. (install_octopi.sh)

supernielsen, 2019-06-20:
That i did try and it didnt make Any difference wierdly enough as I have
seen that this solved the issue for other people. :)

Arksine, 2019-06-21:
What version of Octopi are you running?

supernielsen, 2019-06-21:
Hello everyone i decided to do it all over and not it works perfectly.. :D

closed 2019-06-26 by klipper-gitissuebot