Klipper3d/klipper #1746: Latest Master LPC Firmware seems to have "issues"

Stephan3, 2019-06-21:
Hello @KevinOConnor.

During my tests with tmc 2209/2208/5160 the last 14 days i had used my forked branch. 
All worked Well two rigs were printing 24/7. During my preperations for the tmc 2209 Pullrequest i merged your latest Master and flashed the SKRs in one of my vorons to the latest firmware.(as i have to du to "protocol mismatch").

I am facing following symptoms:
- If i run a macro which is a long running one like leveling with homing, warmup travel and such and start a sdprint, the mcu is shutting down with:
```
DWC2:96909.86 - 192.168.100.20 - GET - /rr_gcode?gcode=G28
G28
M32 /gcodes/Z_E_FSR_pi_24h_30m_305g.gcode
DWC2:096913.6 - 192.168.100.20 - GET - /rr_gcode?gcode=M32%20%220%3A%2Fgcodes%2FZ_E_FSR_pi_24h_30m_305g.gcode%22
Starting SD card print (position 0)
Unknown command:"M201"
Unknown command:"M203"
Unknown command:"M205"
Unknown command:"M205"
Unknown command:"PRINT_START"
DWC2:96913.72 - 192.168.100.20 - GET - /rr_fileinfo
stepcompress o=5 i=-167287365 c=1 a=0: Invalid sequence
Internal error on command:"G1"
Traceback (most recent call last):
  File "/root/klipper/klippy/gcode.py", line 221, in _process_commands
    handler(params)
  File "/root/klipper/klippy/gcode.py", line 75, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/root/klipper/klippy/gcode.py", line 513, in cmd_G1
    self.move_with_transform(self.last_position, self.speed)
  File "/root/klipper/klippy/extras/bed_mesh.py", line 160, in move
    self.toolhead.move([x, y, z + self.fade_target, e], speed)
  File "/root/klipper/klippy/toolhead.py", line 364, in move
    self.move_queue.add_move(move)
  File "/root/klipper/klippy/toolhead.py", line 193, in add_move
    self.flush(lazy=True)
  File "/root/klipper/klippy/toolhead.py", line 181, in flush
    move.move()
  File "/root/klipper/klippy/toolhead.py", line 106, in move
    self.toolhead.kin.move(next_move_time, self)
  File "/root/klipper/klippy/kinematics/corexy.py", line 116, in move
    rail_z.step_itersolve(cmove)
  File "/root/klipper/klippy/stepper.py", line 199, in step_multi_itersolve
    stepper.step_itersolve(cmove)
  File "/root/klipper/klippy/mcu.py", line 134, in step_itersolve
    raise error("Internal error in stepcompress")
error: Internal error in stepcompress
Dumping gcode input 2 blocks
Read 572.140520: 'dump_tmc stepper=extruder\n'
Read 945.141740: 'dump_tmc stepper=extruder\n'
gcode state: absolute_coord=True absolute_extrude=False base_position=[0.0, 0.0, 0.0, 0.0] last_position=[279.093, 7.995, 0.2, 1756.01335] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.0166666666667 extrude_factor=1.0 speed=20.0
Virtual sdcard (162): 'l extrusion width = 0.54mm\n; solid infill extrusion width = 0.42mm\n; top infill extrusion width = 0.36mm\n; first layer extrusion width = 0.40mm\n\nM201 X4000 Y4000 Z500 E10000 ; sets maximum accelerations, mm/sec^2\nM203 X500 Y500 Z25 E160 ; sets maximum feedrates, mm/sec\nM204 P4000 R4000 T4000 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2\nM205 X10.00 Y10.00 Z0.20 E2.50 ; sets the jerk limits, mm/sec\nM205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec\nM107\nM190 S110 ; set bed temperature and wait for it to be reached\nM104 S250 ; set temperature\nprint_start\n; Filament gcode\nM109 S250 ; set temperature and wait for it to be reached\nG21 ; set units to millimeters\nG90 ; use absolute coordinates\nM83 ; use relative distances for extrusion\nG1 Z0.200 F12000.000\nG1 E-6.00000 F4800.00000\nG1 Z0.600 F12000.000\nG1 X11.285 Y9.908\nG1 Z0.200\nG1 E6.00000 F4800.00000\nM204 S4000\nG1 F1200\nG1 X12.410 Y8.970 E0.03828\nG1 X13.900 Y8.169 E0.04417\nG1 X15.321 Y7.744 E0.03876\nG1 X16.784 Y7.595 E0.03843\n'
Upcoming (1186): 'G1 X279.093 Y7.995 E6.85371\nG1 X280.138 Y8.064 E0.02736\nG1 X281.044 Y8.234 E0.02408\nG1 X282.297 Y8.634 E0.03437\nG1 X283.486 Y9.2'
Internal error on command:"G1"
virtual_sdcard dispatch
Traceback (most recent call last):
  File "/root/klipper/klippy/extras/virtual_sdcard.py", line 180, in work_handler
    self.gcode.run_script(lines[-1])
  File "/root/klipper/klippy/gcode.py", line 289, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/root/klipper/klippy/gcode.py", line 221, in _process_commands
    handler(params)
  File "/root/klipper/klippy/gcode.py", line 75, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/root/klipper/klippy/gcode.py", line 513, in cmd_G1
    self.move_with_transform(self.last_position, self.speed)
  File "/root/klipper/klippy/extras/bed_mesh.py", line 160, in move
    self.toolhead.move([x, y, z + self.fade_target, e], speed)
  File "/root/klipper/klippy/toolhead.py", line 364, in move
    self.move_queue.add_move(move)
  File "/root/klipper/klippy/toolhead.py", line 193, in add_move
    self.flush(lazy=True)
[... 470 lines cut ...]
  File "/root/klipper/klippy/homing.py", line 118, in home_rails
    self.homing_move(movepos, endstops, homing_speed, dwell_t=dwell_t)
  File "/root/klipper/klippy/homing.py", line 71, in homing_move
    mcu_endstop.home_wait(move_end_print_time)
  File "/root/klipper/klippy/mcu.py", line 199, in home_wait
    while self._check_busy(eventtime, home_end_time):
  File "/root/klipper/klippy/mcu.py", line 214, in _check_busy
    s.note_homing_end(did_trigger=True)
  File "/root/klipper/klippy/mcu.py", line 125, in note_homing_end
    [self._oid], response='stepper_position', response_oid=self._oid)
  File "/root/klipper/klippy/serialhdl.py", line 188, in send_with_response
    return src.get_response()
  File "/root/klipper/klippy/serialhdl.py", line 230, in get_response
    raise error("Timeout on wait for '%s' response" % (self.name,))
error: Timeout on wait for 'stepper_position' response
Internal error on command:"G28"
failed: G28Timeout on wait for 'stepper_position' response
```

I cant tell since which change this happens as my last merge is about ~1,5 month old. For me this random feature is new. I ran theese macros since ages.

BR
Stephan

[klippy.log](https://github.com/KevinOConnor/klipper/files/3314092/klippy.log)

KevinOConnor, 2019-06-21:
It sounds like a bad merge.  If you can reproduce with the pristine Klipper code, I'll take a closer look.

-Kevin

KevinOConnor, 2019-06-28:
Any further updates on this?

-Kevin

Stephan3, 2019-06-28:
havent the time yet to update my rigs to master. I will try on sunday.

KevinOConnor, 2019-07-10:
I'm going to close this for now - we can reopen if it is reproducible on the pristine Klipper code.

-Kevin

closed 2019-07-10 by KevinOConnor